rust-osdev/x86_64 #7: Bitflags will clear unsupported fields.

le-jzr, 2017-08-28:
Currently, unknown flags in bitfields (e.g. `cr4`) get automatically cleared by the library. This means that if part of the kernel uses a flag this library is not yet aware of, changing flags with this library will corrupt the environment.

This doesn't matter if all parts of the kernel use (the same version of) this crate to access the registers, but that doesn't always need to be the case. I think it would be safer to make sure unknown flags are kept unchanged (by adding a mask with all bits set? I'm not sure about that part).

phil-opp, 2017-08-28:
Thanks for reporting! I agree that this is a problem.

So we need to change [Cr4](https://github.com/phil-opp/x86_64/blob/522538c2f1add6a0a8032c1157567f4122e8e59a/src/registers/control_regs.rs#L28) (and the other bitflag types) somehow. We could wrap them in a struct that also saves the unknown flags and ORs both values together before saving: `struct Cr4 { other_flags: usize; flags: Cr4Flags; }`. The `other_flags` field is the result of an XOR of the original value and the result of `from_bits_truncate`. Alternatively we could create our own `bitflags` macro with a `from_bits_ignore` function, which _ignores_ unknown bits instead of dropping them.

le-jzr, 2017-08-28:
Alternatively, we can just make bitflags consider all bits relevant, like this:
```rust
pub flags Cr4: usize {
    // ...
    // ... supported flags
    // ...

    const ALL = core::usize::MAX,
}
```

phil-opp, 2020-04-10:
I think this no longer applies today, therefore closing.

closed 2020-04-10 by phil-opp