rust-osdev/x86_64 #8: Mistake in ports asm

vlad9486, 2017-10-28:
The clobber is "edi" but used "esi".
Here [https://github.com/phil-opp/x86_64/blob/master/src/instructions/port.rs#L43]() and two more lines.

phil-opp, 2017-10-28:
Thanks for reporting!

This assembly line comes from [gz/rust-x86](https://github.com/gz/rust-x86), which took it from [Tobba/libcpu](https://github.com/Tobba/libcpu). I don't know the exact requirements for inline assembly (e.g. do I need to add input registers to the clobber list), but I try to find it out. 

Maybe @Tobba, @gz, or @Ericson2314 know more?

gz, 2017-11-30:
It's usually additional registers which get changed apart from the ones declared in output (i.e., some may change as a result of side-effects which the compiler can't figure out)

vlad9486, 2017-12-05:
@phil-opp  You are right. Looks like the code is ok.

phil-opp, 2017-12-05:
Ok, thanks for reporting anyway! It's always good to double check inline assembly.

closed 2017-12-05 by phil-opp