rust-osdev/x86_64 #9: Halt function doesn't return !/isn't diverging

Restioson, 2017-11-30:
Logically, the `halt` instruction can never return. Thus, it should be marked as such by declaring it as diverging (returning `!`). Since the compiler doesn't know that the `asm` block itself is diverging, one could add a `loop {}` at the end of the function to trick it.

phil-opp, 2017-11-30:
> Logically, the halt instruction can never return.

This is not quite true. From wikipedia: "_In the x86 computer architecture, HLT (halt) is an assembly language instruction which halts the central processing unit (CPU) **until the next external interrupt is fired**_". So the function can and will return, for example on every timer interrupt.

Restioson, 2017-11-30:
Hmm, ok. I wasn't aware of that. Thanks! Should I close?

phil-opp, 2017-11-30:
Yeah, I think we can close this. Thanks for opening this issue!

closed 2017-11-30 by phil-opp