rust-osdev/x86_64 #10: Registering random interrupt handlers in the IDT causes exception 0xb.

too-r, 2018-01-07:
Hello,

I've been trying to setup system calls in my kernel, and I noticed a weird bug.

```rust
//...
idt.interrupts[0x80].set_handler_fn(syscall::syscall_handler);
```
When trying to trigger a test system call, the system experienced a double fault. After a bit of debugging, I realised that the first exception was a `Segment Not Present` exception. The instruction pointer pointed to the exact instruction `int 0x80`. This led me to believe that somehow the present bit was being unset. I tested this theory by cloning the `blog_os` repo, and registering a dummy handler on the same vector `0x80`. Attempting to trigger a test interrupt caused the same `Segment Not Present` exception as I had experienced in my own kernel. I wonder if you can shed some light on this. I have a suspicion that this might be related to #7, but I'm not certain.

Thanks.

phil-opp, 2018-01-07:
Hmm, what is the error code of the exception? It should be the selector of the present entry.

too-r, 2018-01-07:
The error code as given by the function argument is `0x802`. The exception stack, as might be expected, points to the code segment of the GDT.

phil-opp, 2018-01-07:
Ok, I think I found the problem. The IDT struct looks like this:

```
pub struct Idt {
    pub divide_by_zero: IdtEntry<HandlerFunc>,
    pub debug: IdtEntry<HandlerFunc>,
    […]
    pub interrupts: [IdtEntry<HandlerFunc>; 224],
    // some fields omitted
}
```

So we have special cases for the first 32 interrupts, which means that the `interrupts` field starts at index 32. So try doing `idt.interrupts[0x80 - 32].set_handler_fn(syscall::syscall_handler)` instead.

This seems like a huge pitfall, so I think we should look for a better solution, or at least document it explicitly.

too-r, 2018-01-07:
This works, thanks. Weird that this has to be done though. Maybe we could modify the Index implementations to account for this? As a side note: In my kernel I have remapped the PICs to `0x20` and `0x28`, and have no trouble thus registering the timer and keyboard IRQs to the first and second indices of the `interrupts` field. I'm assuming this bug won't affect IRQs given that the PIC sends interrupts to the CPU separately.

phil-opp, 2018-01-07:
> This works, thanks.

Great!

> Weird that this has to be done though. Maybe we could modify the Index implementations to account for this?

The problem is that we want to special case interrupts 0-32, since they have fixed meaning and some of them have an additional error code (which means they have a different type). If we would change the index implementation of the `interrupts` field, then we would have to issue a runtime panic for indices smaller than 32, which is not ideal either (but maybe better?).

> As a side note: In my kernel I have remapped the PICs to 0x20 and 0x28, and have no trouble thus registering the timer and keyboard IRQs to the first and second indices of the interrupts field. I'm assuming this bug won't affect IRQs given that the PIC sends interrupts to the CPU separately.

This "affects" neither software nor hardware interrupts, it's just a different numbering convention. So your `int 0x80` goes to the `0x80`s entry in the IDT, which is the `0x80-32`s entry in the `interrupts` array. Equally, your timer interrupt goes to entry `0x20` (due to the PIC remapping), which is the `0x20-32=0`s entry in the interrupt array.

too-r, 2018-01-07:
Ah I see, makes sense. Thanks.

closed 2018-03-12 by too-r

JoshMcguigan, 2019-06-24:
Note to future readers:

The interrupts field on the Idt struct is no longer public, and an Index and IndexMut impl was added for Idt. This means `idt.interrupts[0x80 - 32].set_handler_fn(syscall::syscall_handler)` should now be expressed as `idt[0x80].set_handler_fn(syscall::syscall_handler)`.