rust-osdev/x86_64 #21: How to use From trait to convert os_bootinfo::FrameRange into PhysFrameRange

robert-w-gries, 2018-04-17:
This issue bugged me over the weekend.  I tried to convert an `os_bootinfo::FrameRange` struct to `PhysFrameRange` using the `From` trait and saw a compilation error.  I tried `cargo clean` and made sure all crates were up-to-date, and I still couldn't get it working.

I have a sinking feeling that this is a simple mistake on my end, but I wasn't able to search for a similar issue.

### Code

```rust
let mut range: PhysFrameRange = PhysFrameRange::from(frame_range);
```

### Error

The error message doesn't seem particuarly helpful.

```
error[E0277]: the trait bound `x86_64::structures::paging::PhysFrameRange<_>: core::convert::From<os_bootinfo::FrameRange>` is not satisfied
  --> src/arch/x86/memory/area_frame_allocator.rs:36:41
   |
36 |         let mut range: PhysFrameRange = PhysFrameRange::from(frame_range);
   |                                         ^^^^^^^^^^^^^^^^^^^^ the trait `core::convert::From<os_bootinfo::FrameRange>` is not implemented for `x86_64::structures::paging::PhysFrameRange<_>`
   |
   = help: the following implementations were found:
             <x86_64::structures::paging::PhysFrameRange as core::convert::From<os_bootinfo::memory_map::FrameRange>>
   = note: required by `core::convert::From::from`
```

### Other attempts

I tried passing a size type with the `PhysFrameRange` creation, and I still couldn't get it working.

```rust
let mut range: PhysFrameRange = PhysFrameRange::<Size4KB>::from(frame_range);
```

```
error[E0277]: the trait bound `x86_64::structures::paging::PhysFrameRange: core::convert::From<os_bootinfo::FrameRange>` is not satisfied
  --> src/arch/x86/memory/area_frame_allocator.rs:36:41
   |
36 |         let mut range: PhysFrameRange = PhysFrameRange::<Size4KB>::from(frame_range);
   |                                         ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^ the trait `core::convert::From<os_bootinfo::FrameRange>` is not implemented for `x86_64::structures::paging::PhysFrameRange`
   |
   = help: the following implementations were found:
             <x86_64::structures::paging::PhysFrameRange as core::convert::From<os_bootinfo::memory_map::FrameRange>>
   = note: required by `core::convert::From::from`
```

phil-opp, 2018-04-17:
That's a strange error. My only idea is that this could be a version conflict, i.e. the from trait is implemented for a different version than you passed. X86_64 uses os_bootinfo version 0.2.0, maybe you're still using the alpha version?

robert-w-gries, 2018-04-18:
Changing my `Cargo.toml` to the following worked:

```diff
-os_bootinfo = { git = "https://github.com/rust-osdev/os_bootinfo" }
+os_bootinfo = "0.2.0"
```

I previously had `x86_64`, `os_bootinfo`, and `bootloader` crates pointing to the latest git revision so I could avoid using out-of-date crates 😄 

This requirement is rough if you want to use the latest `os_bootinfo` on Github or use a local `os_bootinfo` repository.

phil-opp, 2018-04-18:
> This requirement is rough if you want to use the latest os_bootinfo on Github or use a local os_bootinfo repository.

The problem is that we need to implement the From trait somewhere. We can't implement it in os_bootinfo, since we want to avoid any dependencies for that crate. And even if we implemented it in os_bootinfo, we would have the same problem just with different x86_64 versions.

Also, it's dangerous to use different os_bootinfo versions because they are passed over a C API, which is not type safe. If the bootloader uses a different version than the kernel, you might only get junk values. So our idea was to keep os_bootinfo as small as possible (e.g. no dependencies), so that it's as stable as possible. This means that there shouldn't be many changes to os_bootinfo in the future (maybe a few changes are still needed until we finalized the design).

phil-opp, 2018-05-07:
@robert-w-gries I think this is solved. Can we close this issue?

closed 2018-05-07 by robert-w-gries