rust-osdev/x86_64 #28: GS_BASE duplication

4e554c4c, 2018-05-29:
https://github.com/rust-osdev/x86_64/blob/ed1cf71fc821ab2630849838a0008fa5d1fad41f/src/registers/msr.rs#L3128-L3132
is this intended?

lachlansneff, 2018-05-30:
The `registers/msr.rs` file is from when this crate was just called `x86`. No one has looked at it in a looooong time. There are presumably duplicates and possibly incorrect values.

phil-opp, 2018-06-03:
We're going to remove the `msr` module for the upcoming 0.2.0 release, because it is too cluttered and difficult to maintain. 

We still provide a `registers::model_specific` module for accessing MSRs, but no longer the complete set of base addresses. We're happy to merge pull requests that re-add individual registers such as GS_BASE that are actually needed.

closed 2018-06-03 by phil-opp

4e554c4c, 2018-06-03:
Sounds good, thanks :)