rust-osdev/x86_64 #31: Beta testing 0.2.0

phil-opp, 2018-06-09:
Yesterday I released version `0.2.0-beta` of `x86_64`. There have been [a lot of changes](https://github.com/rust-osdev/x86_64/compare/v0.1.2...master) since version `0.1.2`, some of them breaking.

The new version is a bit stripped down in some areas to keep it maintainable. Please let me know if you are missing any functionality. There are also a few new modules, the most significant beeing the new abstractions for recursive paging.

The crate is [fully documented](https://docs.rs/x86_64/0.2.0-beta/x86_64/) now and we have a CI job that checks that no undocumented items exist and also that no warnings occur.

cc @lachlansneff @4e554c4c @Restioson @gegy1000 @robert-w-gries

lachlansneff, 2018-06-09:
Is this any different than the most recent alpha?

phil-opp, 2018-06-09:
Yes, there are [lots of changes](https://github.com/rust-osdev/x86_64/compare/v0.2.0-alpha-020...v0.2.0-beta).

phil-opp, 2018-06-09:
(There might be some more changes to the recursive paging module as part of #30.)

lachlansneff, 2018-06-11:
Switched Nebulet over to `x86_64` `0.2.0-beta` and it works just fine

phil-opp, 2018-06-11:
Perfect, thanks for testing!

phil-opp, 2018-06-14:
Version 0.2.0 was released in 2a40598dd66714966d7ff19cbd5c278e2021cfec.

closed 2018-06-14 by phil-opp