rust-osdev/x86_64 #44: Port IO sizes

TyPR124, 2018-10-08:
This doesn't look right to me, but I could be misunderstanding something.

In  x86_64/src/instructions/port.rs

```
impl PortReadWrite for u16 {
    #[inline]
    unsafe fn read_from_port(port: u16) -> u16 {
        let value: u16;
        asm!("inw %dx, %ax" : "={ax}"(value) : "{dx}"(port) :: "volatile");
        value
    }

    #[inline]
    unsafe fn write_to_port(port: u16, value: u16) {
        asm!("outw %ax, %dx" :: "{dx}"(port), "{al}"(value) :: "volatile");
    }
}

impl PortReadWrite for u32 {
    #[inline]
    unsafe fn read_from_port(port: u16) -> u32 {
        let value: u32;
        asm!("inl %dx, %eax" : "={ax}"(value) : "{dx}"(port) :: "volatile");
        value
    }

    #[inline]
    unsafe fn write_to_port(port: u16, value: u32) {
        asm!("outl %eax, %dx" :: "{dx}"(port), "{al}"(value) :: "volatile");
    }
}
```

Shouldn't write for u16 be:
 `asm!("outw %ax, %dx" :: "{dx}"(port), "{ax}"(value) :: "volatile");`,
and for u32, read:
`asm!("inl %dx, %eax" : "={eax}"(value) : "{dx}"(port) :: "volatile");`
and write:
`asm!("outl %eax, %dx" :: "{dx}"(port), "{eax}"(value) :: "volatile");` ?

phil-opp, 2018-10-09:
Yeah, I think these are probably copy&paste bugs. Thanks a lot for reporting! Could you file a pull request for this?

TyPR124, 2018-10-10:
Fixed ihn #45

closed 2018-10-10 by TyPR124