rust-osdev/x86_64 #56: Support RdRand

kamirr, 2019-01-30:
Hello there,
It would be great if there was an `rdrand` function in the `instructions` submodule for seeding RNGs. It's relatively simple to implement and I believe many people would find it useful.

What are your thoughts on this?

phil-opp, 2019-01-31:
Sure, sounds useful! I'm happy to merge a pull request that implements it.

kamirr, 2019-01-31:
I implemented the first one in my PR as it was pretty simple and lets me experiment a bit with this concept. Feel free to try it out.

closed 2019-02-03 by kamirr

TyPR124, 2019-02-06:
Hey, I like this and may use it, but I'm wondering if we should be checking for a valid return? Per the link

https://www.felixcloutier.com/x86/rdrand (and also the referenced Intel doc)

the instruction can fail. It sets CF to 1 on success, and 0 on failure. The link says it writes 0 to the destination, but the Intel doc just says it may "return no data"

One option I can think of to deal with this is to return `Option<NonZeroU64>` from `get`, which would interpret a 0 value as `None` and allow the caller to deal with a failure scenario. Personally, if doing this, I think it'd be a good idea to set rax to 0 before rdrand, to be sure it isn't leaving a stale (non-random) value in rax.

phil-opp, 2019-02-06:
@TyPR124 Oh, I completely missed that! We should definitely check for failure. Do you mind creating a PR for this?

kamirr, 2019-02-06:
Huh, no idea how I missed that. I don't think using `Option<NonZeroU64>` is a good idea though, `0` is technically a valid result so this approach wouldn't be strictly correct.

Considering that this `RDRAND` is slow anyway, I think that we could afford to return normal `Option<u64>`.

TyPR124, 2019-02-06:
I can submit a PR, but it might take me a couple of days to get to it.

@kamirr You're right, 0 is a valid value. I was already hesitant about it for various reasons, that being one of them. On second thought, Option<u64> seems like the best thing to do. If you want to implement this, feel free. As I said, I may not get to it right away.

Taking this a bit further, RdRand (and RdSeed, if we wanted to implement that) can also return u32 or u16. And I think it would technically be okay to split a u16 in half to return a u8, but I'm not sure how I feel about that.

I can work on generalizing this a bit as well if it is wanted.

phil-opp, 2019-02-07:
Ok, so I think we should just [check the carry flag](https://stackoverflow.com/questions/3139772/check-if-carry-flag-is-set) after executing `rdrand` and pass it from inline assembly to Rust through an additional output parameter.

> Taking this a bit further, RdRand (and RdSeed, if we wanted to implement that) can also return u32 or u16. And I think it would technically be okay to split a u16 in half to return a u8, but I'm not sure how I feel about that.
> 
> I can work on generalizing this a bit as well if it is wanted.

We should introduce separate `rdrand` versions that use the 32/16-bit registers directly. This way we don't needlessly throw away any random bits.

TyPR124, 2019-02-08:
@kamirr I submitted a PR for this if you want to take a look