rust-osdev/x86_64 #65: Create attribute #![feature(try_from)] required

subsid, 2019-03-20:
I was trying to use the new [Paging Implementation](https://github.com/phil-opp/blog_os/tree/post-10) post and got the following error from this crate:

```
error[E0658]: use of unstable library feature 'try_from' (see issue #33417)
 --> /Users/subsid/.cargo/registry/src/github.com-1ecc6299db9ec823/x86_64-0.5.2/src/addr.rs:1:27
  |
1 | use core::convert::{Into, TryInto};
  |                           ^^^^^^^
  |
  = help: add #![feature(try_from)] to the crate attributes to enable
```
Adding `#![feature(abi_x86_interrupt)]` to this crate's lib.rs solved the issue.

Thanks!

AntoineSebert, 2019-03-20:
Isn't `try_from` part of the standard library now ?
Have you tried to update your Rust toolchain ?

subsid, 2019-03-20:
@AntoineSebert oops. Yes, thank you!

closed 2019-03-20 by subsid

AntoineSebert, 2019-03-20:
Master the jedi arts of updating toolchains, you need, young padawan.