rust-osdev/x86_64 #90: Reduce compile times by removing dependencies

phil-opp, 2019-09-25:
I measured the compile time of the blog_os project with the new `-Z timings` flag:

[image]

The `x86_64` crate and especially its `ux` and `array-init` dependencies take very long to compile. Maybe it's possible to remove or replace these dependencies somehow.

phil-opp, 2019-09-25:
I opened #91 to evaluate the compile times without `ux`. New timings:

[image]

m-ou-se, 2020-02-10:
Since  #116, `array-init` is also gone. Only three dependencies are left:

- `bit_field` with no dependencies,
- `bitflags` with no dependencies, and
- `cast` with a build-dependencies: `rustc_version` -> `semver` -> `semver_parser`.

phil-opp, 2020-02-10:
With `cast` removed in #124, I think we can close this issue given that the two remaining dependencies are fairly lightweight.

closed 2020-02-10 by phil-opp