rust-osdev/x86_64 #120: GlobalDescriptorTable::add_entry creates SegmentSelector with hard coded privilege level

haraldh, 2020-01-24:
https://github.com/rust-osdev/x86_64/blob/99b8a0af46300ee99411b59ed0f47ff599748c15/src/structures/gdt.rs#L110

Why is the privilege level hard coded?

phil-opp, 2020-01-24:
Good question… I think the idea was to set the requested privilege level to Ring0 by default so that all allowed segments can be accessed with the descriptor. However, I'm not sure about this anymore. What would be your preferred solution?

Independent of this, we should add a `set_rpl` method to the `SegmentSelector` type.

phil-opp, 2020-02-10:
> Independent of this, we should add a `set_rpl` method to the `SegmentSelector` type.

This was added as part of #119. So now it is at least possible to change the RPL of a segment selector.

haraldh, 2020-07-03:
https://github.com/rust-osdev/x86_64/pull/153 fixed this .. thanks

closed 2020-07-03 by haraldh