rust-osdev/x86_64 #125: #[inline] all the things?

m-ou-se, 2020-02-10:
There are lots of trivial and very small functions in this crate that could (should?) be marked `#[inline]`. E.g. `VirtAddr::p1_index`, Add(Assign)/Sub(Assign) implementations, `PrivilegeLevel::from_u16`, etc. etc.

Should all of them be marked `#[inline]`? Or should users of this crate always depend on LTO?

kivimango, 2020-02-21:
Second answer

https://internals.rust-lang.org/t/when-should-i-use-inline/598

AntoineSebert, 2020-02-21:
I have followed the link and read the thread, but I still have a question regarding the `#[inline]` attribute. In C++ an equivalent keyword also exists, but it's merely a hint for the compiler. Does Rust has a similar approach ?

phil-opp, 2020-02-25:
As far as I know, the `#[inline]` attribute is just a hint for the compiler that it should consider this function for cross-crate inlining. Without the attribute, inlining across crate bounds is only performed when link time optimization is enabled.

The `#[inline]` attribute is different from the `#[inline(always)]` attribute, which always forces inlining and is generally discouraged (because the compiler usually knows it better).

> Should all of them be marked `#[inline]`? Or should users of this crate always depend on LTO?

Forcing the users to always use LTO does not seem like the best solution. So I think it's fine to add `#[inline]` to all small functions of this crate and I'm happy to merge a PR for this.

AntoineSebert, 2020-02-25:
I have done #129 .
I'm not 100% sure of my modifications, hopefully it's just a hint for the compiler, so the observable behaviour should not change.
Tested with `cargo build` and `cargo test`.
Did I miss something ?

phil-opp, 2020-02-26:
Fixed in #129.

closed 2020-02-26 by phil-opp