rust-osdev/x86_64 #131: `InterruptStackFrame` and `unsafe` semantics

AntoineSebert, 2020-02-26:
The struct `InterruptStackFrame` implemements a function `as_mut()` as unsafe (https://github.com/rust-osdev/x86_64/blob/master/src/structures/idt.rs#L763), [claiming](https://github.com/rust-osdev/x86_64/blob/master/src/structures/idt.rs#L758-L759):

> /// This function is unsafe since modifying the content of the interrupt stack frame
/// can easily lead to undefined behavior.

However the documentation at https://doc.rust-lang.org/std/keyword.unsafe.html specifies:

> Code or interfaces whose memory safety cannot be verified by the type system.

Since the type system can verify the function's memory safety, I suggest to remove the `unsafe` keyword.

phil-opp, 2020-02-26:
While the creation of the `&mut` reference is safe, you can do very unsafe things with this reference. For example, you can set the instruction pointer to an arbitrary address, causing the CPU to interpret random bytes as code when the interrupt handler returns.

So the safety of the interface provided by the `&mut` reference cannot be verified by the type system. As such, the usage of this interface must be unsafe. This is ensured by the unsafe `as_mut` method because it makes the creation of the interface unsafe.

AntoineSebert, 2020-02-26:
Oh okay.
I opened this issue also because I tried to implement the trait `core::convert::AsMut`, but couldn't because `core::convert::AsMut::as_mut` is not unsafe.
Is there any workaround ?
Because for the type system, `InterruptStackFrame` does not really implement `core::convert::AsMut`, right ?

phil-opp, 2020-02-26:
Why do you want to implement this trait for `InterruptStackFrame`? Just for convenience or is there some other reason?

AntoineSebert, 2020-02-27:
For convenience and to learn, so it's okay I things stay the way they are.

closed 2020-02-27 by AntoineSebert

phil-opp, 2020-02-27:
Thanks for clarifying! We don't really want the convenience here because unsafe operations should always be explicit.