rust-osdev/x86_64 #139: Provide convenience methods to create PageTables

vinaychandra, 2020-03-13:
The current implementations of offset and recursive page table require that `new` to be called with an already existing page table. Provide convenience methods to create more page tables from an existing page table or by taking limits as a parameter to the `new` function.

phil-opp, 2020-03-13:
Could you clarify what you mean with "creating more page tables from an existing page table"? Do you mean to create new independent 4-level page table hierarchies for creating other processes?

> by taking limits as a parameter to the `new` function

What kind of limits do you mean?

vinaychandra, 2020-03-13:
Yes, for creating other processes. The limits in sense, for offset page table, the offset and max address may be for example.

phil-opp, 2020-03-13:
Ok, thanks for clarifying! I'm still not sure how such a function should look though. The problem is that the `OffsetPageTable` and the other types only store a `&mut PageTable` reference internally, and thus do not have ownership of the `PageTable` instance. We designed it this way to support both static page tables and heap allocated page tables.

One possible way for creating a new page table hierarchy could be:

```rust
let level_4_page_table = Box::leak(Box::new(PageTable::new()));
let mapper = OffsetPageTable::new(level_4_page_table);
```

This approach works, but it does not handle deallocation of the page table. So you need to do `Box::from_raw` with the level_4_page_table pointer before dropping the process. (You will also need to free all page table frames. We don't have a good solution for that yet.)

vinaychandra, 2020-03-14:
Calling `OffsetPageTable::new` doesn't update the page table passed to it. So, we need to update the page table as well (Please correct me if i am wrong).

phil-opp, 2020-03-14:
What do you mean with "update"?

vinaychandra, 2020-03-14:
Creating entries in the page table. Does calling new automatically create the entries? AFAIK it doesn't.

phil-opp, 2020-03-16:
Ah, I think I understand what you mean now. You basically propose to add methods like the following:

```rust
impl OffsetPageTable {
    /// Creates a new level 4 page table that includes the physical memory mapping.
    ///
    /// Returns a tuple of the new `PageTable` and the offset of the physical memory
    /// mapping. These values can then be passed to the `OffsetPageTable::new`
    /// function.
    pub fn new_level_4_table(&self, frame_allocator: A) -> (PageTable, usize) {…}
}

impl RecursivePageTable {
    /// Creates a new level 4 page table with a single recursive entry.
    ///
    /// Returns a tuple of the new `PageTable` and the index of the recursive entry.
    /// These values can then be passed to the `RecursivePageTable::new` function.
    pub fn new_level_4_table(&self) -> (PageTable, PageTableIndex) {…}
}
```

Did I get understand that right?

Such methods seem like a useful addition!

vinaychandra, 2020-03-16:
That is correct. It could also be done so that we pass in the offset instead of it auto deciding an offset but that's optional.

phil-opp, 2020-03-16:
Oh yeah, that makes sense!