rust-osdev/x86_64 #149: Allow reading present flags for a PageTable Entry

tscs37, 2020-04-11:
It would be convenient if there is a method to translate a virtual address to the corresping PageTableFlags, this way the interrupt handler can check which flags are set on the entry and handle faults accordingly.

At the moment, the flags are not easily accessible from a pagetable, which makes it hard to debug issues where the active pagetable flags are unclear.

phil-opp, 2020-04-14:
Just to ensure that I understand correctly: You propose a `Mapper` method like the following:

```rust
fn flags_for_addr(&self, addr: VirtAddr) -> Option<PageTableFlags> {…}
```
?

tscs37, 2020-04-15:
Essentially, yes.

phil-opp, 2020-04-15:
Thanks for clarifying! I agree that such a function would be useful. Perhaps it's also a good idea to provide a general `translate` method that yields both the mapped physical address and the flags for the mapping. This way, only a single lookup would be required if both information are needed.

mental32, 2020-12-28:
Should this issue be closed on account of #207 being merged?

phil-opp, 2020-12-28:
Yes, I think we can close this. Thanks!

closed 2020-12-28 by phil-opp