rust-osdev/x86_64 #152: Rename `MapperAllSizes` to `Translate`?

phil-opp, 2020-04-23:
The [`MapperAllSizes`](https://docs.rs/x86_64/0.10.1/x86_64/structures/paging/mapper/trait.MapperAllSizes.html) trait currently provides only methods for translating addresses. While it is currently a supertrait of `Mapper<{Size4KiB, Size2MiB, Size1GiB}`, this isn't really required for any of its methods. It might thus be a good idea to remove the supertrait relationship and rename the trait to e.g. `Translate`.

phil-opp, 2020-10-18:
It might be worth keeping `MapperAllSizes` as an alias for `Mapper<{Size4KiB, Size2MiB, Size1GiB}` (without any additional methods).

mental32, 2020-12-28:
Triage: should this be closed on account of #211 ?

phil-opp, 2020-12-28:
Yes, thanks!

PR #211 (2020-12-28): Restructure the `TranslateResult` type and create separate `Translate` trait
Implements the design proposed in https://github.com/rust-osdev/x86_64/pull/150#issuecomment-618342192 and fixes #152.

This is a **breaking change**: The `MapperAllSizes` trait no longer has any methods and is only kept for convenience. The `translate` and `translate_addr` methods are now available through a new `Translate` trait. The `TranslateResult` type was restructed to avoid repititions.
closed 2020-12-28 by phil-opp