rust-osdev/x86_64 #160: No kernel data segment?

samueltardieu, 2020-06-16:
There seems to be no `Descriptor::kernel_data_segment()`. If I understand correctly, the presence of such a segment is expected in `IA32_STAR` when doing a `syscall`, as it is used to load the `SS` register. Or did I misunderstand the x86_64/amd64 manual?

If needed, I'd be happy to propose a PR.

xfoxfu, 2020-06-19:
In [GlobalDescriptorTable document](https://docs.rs/x86_64/0.11.0/x86_64/structures/gdt/struct.GlobalDescriptorTable.html), there says _Data segment registers in ring 0 can be loaded with the null segment selector_. So I think using just the null segment selector maybe is ok.

samueltardieu, 2020-06-19:
Unfortunately when doing a `syscall` the data selector is necessarily the one after the code selector (there is only one offset for both kernel code and data segment in register `IA32_STAR`), so I cannot use the null segment selector which is at offset 0.

Note that even the code to install this offset into `IA32_STAR` checks that the data selector is located right after the code selector: https://github.com/rust-osdev/x86_64/blob/master/src/registers/model_specific.rs#L339-L341

phil-opp, 2020-10-04:
This seems to be resolved by https://github.com/rust-osdev/x86_64/pull/181. It introduced a new [`kernel_data_segment`](https://docs.rs/x86_64/0.12.2/x86_64/structures/gdt/enum.Descriptor.html#method.kernel_data_segment) function.

closed 2020-10-04 by phil-opp