rust-osdev/x86_64 #164: New assembly syntax

rybot666, 2020-06-24:
Our crate is currently broken on nightly because we need to update our assembly syntax or keep using legacy assembly. I will get around to this if nobody else does soon. This will break compatibility with nightlies older than a few weeks at maximum, so this will probably be a breaking change (@phil-opp please confirm this).

phil-opp, 2020-06-24:
I don't think that anything is broken right now, the CI jobs succeeded with the latest nightly (apart from clippy): https://github.com/rust-osdev/x86_64/actions/runs/145726670 . We're using `llvm_asm!` instead of `asm!` since version 0.10.2, perhaps you're on an older version?

I'm of course open to switching to the new `asm!` macro anyway, so feel free to send in a PR if you like. We can publish it as a semver-breaking version to avoid breaking people on older nightlies if we want to.

rybot666, 2020-06-27:
Huh, I could have sworn I saw the latest build being broken because of this when I posted it. My bad. Regardless, moving to the new `asm!` macro would be great

PR #165 (2020-09-30): Use new assembly syntax
This pull request updates `x86_64` to use the new `asm!` macro. This will break older nightlies, but makes the syntax easier to read.

Fixes #164
closed 2020-09-30 by josephlr