rust-osdev/x86_64 #171: Descriptor:user_data_segment() does not have Flags::LONG_MODE set

leecannon, 2020-08-02:
The doc-comment of the function states `Creates a segment descriptor for a long mode ring 3 data segment.`

I have not made a pull request to change this as i'm unsure if Flags::LONG_MODE is deliberately not set. My knowledge of the intricacies of x86_64 is sorely lacking.

josephlr, 2020-09-22:
This is intentional. For more information see section 4.8 of [the AMD64 Architecture Programmer’s Manual, Volume 2: System Programming](https://www.amd.com/system/files/TechDocs/24593.pdf).

Basically, in Long mode (which includes 64-bit mode and compatibility mode) Data segments are almost completely ignored. From the manual:
> Segment-limit checking is not performed on any data segments in 64-bit mode, and both the segment-limit field and granularity (G) bit are ignored. The D/B bit is unused in 64-bit mode.
>
> The expand-down (E), writable (W), and accessed (A) type-field attributes are ignored.
>
> A data-segment-descriptor DPL field is ignored in 64-bit mode, and segment-privilege checks are not
performed on data segments. System software can use the page-protection mechanisms to isolate and
protect data from unauthorized access.

In practice this means that a Data Segment (for either the user or the kernel) only needs to set `Flags::USER_SEGMENT ` and `Flags::PRESENT`, and not set `Flags::EXECUTABLE`, every other flag is ignored.

We actually  set `Flags::WRITABLE` and `Flags::DPL_RING_3` unnecessarily which probably confuses things.

josephlr, 2020-09-23:
#181 clarifies some of the documentation here, but I think this can be closed.

phil-opp, 2020-09-23:
Thanks for looking into this @josephlr!

closed 2020-09-23 by phil-opp