rust-osdev/x86_64 #189: Build error on latest nightly

fengyc, 2020-09-30:
os: macos 10.15.6
rust: rustc 1.48.0-nightly (7f7a1cbfd 2020-09-27)

Error message:
```
error[E0635]: unknown feature `const_fn_fn_ptr_basics`
xxx/x86_64-0.12.2/src/lib.rs:7:43
  |
7 | #![cfg_attr(feature = "const_fn", feature(const_fn_fn_ptr_basics))]
  |                                           ^^^^^^^^^^^^^^^^^^^^^^

error: aborting due to previous error

For more information about this error, try `rustc --explain E0635`.
error: could not compile `x86_64`
```

phil-opp, 2020-09-30:
Try updating your Rust nightly. The code was already updated to the very latest nightly, which is rustc 1.48.0-nightly (381b445ff 2020-09-29). If you can't do that,e.g. because you're depending on a component that is not available on the latest nightly, you can temporarily use version 0.12.1 of this crate instead. To do so, you have to change the version requirement to "=0.12.1" (without the `=`, cargo will choose the latest compatible version).

fengyc, 2020-09-30:
The solution works. Thanks.

closed 2020-09-30 by fengyc