rust-osdev/x86_64 #190: interrupts::enable() not working properly

ShardulNalegave, 2020-10-09:
I reading the [Rust OS Dev](https://os.phill-opp.com) tutorial and have been following all the told steps. But I have encountered a problem which is most likely a bug from this crate.
In the [Hardware Interrupts -> Enabling Interrupts](https://os.phil-opp.com/hardware-interrupts/#enabling-interrupts) section we are told to add the following lines:-
```rust
pub fn init() {
    gdt::init();
    interrupts::init_idt();
    unsafe { interrupts::PICS.lock().initialize() };
    x86_64::instructions::interrupts::enable();     // new
}
```

As told next I have also created a handler for timer interrupts. However when I run it, the program behaves in a very weird way, i.e. I am printing some text to the `VGA Text Buffer` but after adding this it doesn't print completely. It is not because anything is panicking and there as no exceptions also! I know this because I am not receiving any Double-Fault exceptions.

After some debugging I found that the problem is with the line `x86_64::instructions::interrupts::enable();`. If I comment it everything works great except of course I don't receive the interrupts. To make sure that this line was the issue I run the program about 10-15 times without this line and it worked as expected but when it is there the issue is present on 99% of the runs.

## Some screenshots of the issue:-
### The desired output =>
[image]

## 2 continuous runs and the results with the interrupts enabled.
### 1. Note that the last line is not printed
[image]
### 2. Only the first line is printed
[image]

**Note:-** As I have told before, I have created an handler for timer events. But, it rarely works. The program runs successfully about once in a 100 tries.

josephlr, 2020-10-09:
Looking at the implementation of `x86_64::instructions::interrupts::enable()` it just calls the [STI instruction](https://www.felixcloutier.com/x86/sti), so I don't think that implementation is incorrect.

It might be better to open this issue with the `blog_os` crate. I would imagine the issue is in how the blog is setting up the IDT or how the handler is implemented.

josephlr, 2020-10-09:
@ShardulNalegave if you open an issue under `blog_os` you should also include the code you are using for your `timer_interrupt_handler`, an incorrect implementation there could be causing your problem.

ShardulNalegave, 2020-10-13:
Hi! Sorry for not replying sooner.
@josephlr Yeah you are right, it looks like the problem is from my side only. As @Caduser2020 had pointed out, this could be a race condition in `vga_println!` and it seems it is. I removed all the `vga_*` code from my timer handler and it works perfectly!
The only problem now is that, I don't know how to fix this issue.... Can someone please help me with it.
@Caduser2020 the code is at https://github.com/ShardulNalegave/catalyst-os

Caduser2020, 2020-10-13:
@ShardulNalegave The issue is [line 108 here](https://github.com/ShardulNalegave/catalyst-os/blob/c3fd0010d0476341ce9ee8604ce96eea2af7b231/src/vga/mod.rs#L108 ). When an interrupt is triggered, the interrupt handler, [here](https://github.com/ShardulNalegave/catalyst-os/blob/c3fd0010d0476341ce9ee8604ce96eea2af7b231/src/interrupts/hardware/timer.rs#L10), calls the `_print` function, which tried to lock the global `WRITER` instance. However, the `WRITER` is already locked in `_start()` by `println!`, causing a deadlock. To fix the deadlock, the `_print` function, [here](https://github.com/ShardulNalegave/catalyst-os/blob/c3fd0010d0476341ce9ee8604ce96eea2af7b231/src/vga/mod.rs#L106), should run `WRITER.lock().write_fmt(args).unwrap();` without interrupts. For example,
```rust
use x86_64::instructions::interrupts;
#[doc(hidden)]
pub fn _print(args: fmt::Arguments) {
    use core::fmt::Write;
    interrupts::without_interrupts(|| {
        WRITER.lock().write_fmt(args).unwrap();
    });
}
```

ShardulNalegave, 2020-10-13:
@Caduser2020 Thanks a lot! That solved the problem.