rust-osdev/x86_64 #193: Tracking issue for breaking changes

josephlr, 2020-10-16:
Many issues/PR currently proposed would break the API of this create. This seems reasonable, but we should try to consolidate changes as much as possible. This issue tracks these changes.

Add any breaking changes you're interested in:
  - [ ] Including `PageTableFlags` in the result of `translate` (https://github.com/rust-osdev/x86_64/pull/150)
  - [ ] Rename `MapperAllSizes` to `Translate` (https://github.com/rust-osdev/x86_64/issues/152)
  - [ ] Potentially changing which methods are required and which are provided for `Mapper` (https://github.com/rust-osdev/x86_64/issues/192)
  - [ ] Remove `PhysFrameRange`, `PhysFrameRangeInclusive`, `PageRange`, `PageRangeInclusive` to instead use the Rust types. 
  - [ ] Make `PhysToVirt` actually map `PhysAddr` to `VirtAddr` (instead of returning a pointer)
  - [ ] Make `DescriptorTablePointer::base` a `VirtAddr` (makes it clear segmentation is applied before paging).
  - [ ] Make `read_rip` return a VirtAddr
  - [ ] Use correct types for `InterruptStackFrameValue`:
    - `code_segment` should be a `SegmentSelector`
    - `cpu_flags` should be a `RFlags`
    - `stack_segment` should be a `SegmentSelector`
  - [ ] Remove deprecated items:
    - `UnusedPhysFrame`
    - `ExceptionStackFrame`
    - `VirtAddr::new_unchecked`

phil-opp, 2020-10-18:
Sounds good to me!

>  Make `PhysToVirt` actually map `PhysAddr` to `VirtAddr` (instead of returning a pointer)

You mean changing the function signature to something like `fn phys_to_virt(&self, phys_addr: PhysAddr) -> VirtAddr`? This would make sense to me. The current signature could become an private function for the `MappedPageTable` type.

phil-opp, 2020-10-30:
(I added https://github.com/rust-osdev/x86_64/pull/199 to the list.)

phil-opp, 2020-12-16:
(I added https://github.com/rust-osdev/x86_64/pull/207 to the list)

phil-opp, 2020-12-28:
I implemented most changes in the `next` branch. The two remaining items are:

- Remove PhysFrameRange, PhysFrameRangeInclusive, etc.
  - Not possible until Step trait is stable
  - Implementing the Step trait on nightly only is possible as a non-breaking change
- Use correct types for InterruptStackFrameValue
  - I'm not sure what to do about the segment selectors, which are pushed as `u64` even though they are only 16 bit

mental32, 2020-12-28:
> I'm not sure what to do about the segment selectors, which are pushed as u64 even though they are only 16 bit

Could a padded newtype work?

```rust
#[repr(C)]
struct PaddedSegmentSelector {
    inner: SegmentSelector,
    _u16: u16,
    _u32: u32,
}
```

phil-opp, 2020-12-28:
Yes, I think that should work.

Given that most of the items from this issue are done, it might make sense to create separate tracking issues for the two remaining items and then close this one.

josephlr, 2021-06-04:
Closing as #262 now tracks the 0.15 release.

closed 2021-06-04 by josephlr