rust-osdev/x86_64 #196: Address is not correctly shifted in page table entries

elegaanz, 2020-10-23:
Hello, I'm not 100% sure of what I'm saying here, so please correct me if I misunderstood something.

Looking at [this](https://wiki.osdev.org/Paging#Page_Table) and [this](https://os.phil-opp.com/paging-introduction/#page-table-format), it looks like the physical address in a page table entry should be shifted by 12 bits to the left (so that it doesn't overlap with the 11 flags bits). However, it looks like `PageTableEntry::set_addr` doesn't do that:

https://github.com/rust-osdev/x86_64/blob/master/src/structures/paging/page_table.rs#L81

It just bit-or the address and the flags, without shifting the address.

If this indeed is a bug, and not just me misunderstanding how this function works, I will happily submit a fix.

josephlr, 2020-10-23:
Thanks for looking at this @elegaanz, the function in this crate is correct.

This is because the physical address is already aligned to 4K bytes (so the lower 12 bits of the physical address are already zero).  The page table entry doesn't shift the physical address, it uses the fact that some bits must be zero, and then uses those bits to store flags.

Sometimes physical page _numbers_ (PPNs) will be used instead of the physical address. For PPNs, you do have to shift left by 12, as the phsyical address `0x12000` has a PPN of `0x12`. 

Does that make sense?

elegaanz, 2020-10-23:
OK, thank you, it makes more sense!

closed 2020-10-23 by elegaanz