rust-osdev/x86_64 #197: What is `UsizeConversions` trait?

toku-sa-n, 2020-10-28:
In `PhysAddr` doc:
>The UsizeConversions trait can be used for performing conversions between u64 and usize.

I googled it but no information about it found. What is this?

phil-opp, 2020-10-28:
See https://docs.rs/usize_conversions/0.2.0/usize_conversions/ 

I don't remember why we're mentioning that crate/trait, probably we re-exported it at some point. So we should probably remove that sentence from the docs.

toku-sa-n, 2020-10-29:
Thank you for your reply. Now that the conversions between `u64` and `usize` can be done with `TryFrom` or `TryInto`, I think just changing `UsizeConversions` to `TryFrom` (or `TryInto`) is enough. If so, I'll create a PR.

phil-opp, 2020-10-29:
Sounds good to me!

PR #198 (2020-10-29): Update docs to suggest `TryFrom` trait
Fixes: #197
closed 2020-10-29 by phil-opp