rust-osdev/x86_64 #202: Is `Mapper::unmap` really safe?

toku-sa-n, 2020-10-31:
I don't think it is. For example, if we unmap the page where the kerenl stack exists, page fault should happen.

phil-opp, 2020-10-31:
The question always is how you define safe and unsafe. For this crate, we use a quite narrow definition of safe. Instead of "cannot cause bad things", we define it as "cannot break memory safety". By this definition, a page fault or even a double/triple fault is not unsafe since no memory is corrupted and no undefined behavior occurs.

The Rust standard library uses a similar definition. This is the reason that functions like [`forget`](https://doc.rust-lang.org/stable/std/mem/fn.forget.html) (leaks memory) or even [`abort`](https://doc.rust-lang.org/stable/std/process/fn.abort.html) (kills the whole process) are safe.

toku-sa-n, 2020-11-03:
I understand. Thank you!

closed 2020-11-03 by toku-sa-n