rust-osdev/x86_64 #229: Write Cr3 with Size1GiB does not work

Luis-Hebendanz, 2021-01-29:
This code fails with:
```rust
    let phys_frame = PhysFrame::<Size1GiB>::from_start_address(PhysAddr::new(p4_addr)).unwrap();
    Cr3::write(phys_frame, Cr3Flags::PAGE_LEVEL_WRITETHROUGH);
```
with
```rust
error[E0308]: mismatched types
   --> src/main.rs:103:16
    |
103 |     Cr3::write(phys_frame, Cr3Flags::PAGE_LEVEL_WRITETHROUGH);
    |                ^^^^^^^^^^ expected enum `x86_64::structures::paging::Size4KiB`, found enum `x86_64::structures::paging::Size1GiB`
    |
    = note: expected struct `PhysFrame<x86_64::structures::paging::Size4KiB>`
               found struct `PhysFrame<x86_64::structures::paging::Size1GiB>`
```

I am not sure why it enforces that type here?

phil-opp, 2021-01-29:
Could you clarify what you're trying to do? A page table is always 4KiB. If you point CR3 to a 1GiB frame, only the first 4KiB would be used, so I'm not sure why one should do this.

Luis-Hebendanz, 2021-01-29:
Hmm yeah initially I was a bit confused but I understood it now. What I don't understand is where this type enforcement comes from. In the source code only the plain PhysFrame is used and there is not mention of the need to use Size4KiB

phil-opp, 2021-01-29:
The type parameter of the `PhysFrame` type is set to `Size4KiB` by default:

https://github.com/rust-osdev/x86_64/blob/8a54bcad4f9f072623e492a337f824320f6812fa/src/structures/paging/frame.rs#L13

So if no explicit generic parameter is mentioned, it defaults to a 4KiB frame.

Luis-Hebendanz, 2021-01-29:
Ohhh I see that makes sense! Thank you :-)

closed 2021-01-29 by Luis-Hebendanz