rust-osdev/x86_64 #231: Interrupt Stack Frame won't print correctly

hamidrezakp, 2021-02-04:
Hi,
I was running post-08 of `blog_os` that noticed the `InterruptStackFrame` type is not printing correctly.
so i checked post-07 and post-09 and the problem was in them too.

Here is the output:
```
InterruptStackFrame {

```
I am running on rust `1.51.0-nightly`, x86_64 `0.13.2`

phil-opp, 2021-02-04:
Thanks for reporting! I'll close this in favor of https://github.com/phil-opp/blog_os/issues/923 and try to look into it soon.

hamidrezakp, 2021-02-04:
btw, they are two different problems, the one in blog_os is raising a double fault and this issue is about not printing correctly.

phil-opp, 2021-02-04:
Ah sorry, thanks for clarifying. Does the printing problem also occur for non-double fault exception? If not, it's probably some kind of deadlock or stack overflow.

hamidrezakp, 2021-02-04:
> Ah sorry, thanks for clarifying. Does the printing problem also occur for non-double fault exception? If not, it's probably some kind of deadlock or stack overflow.

I checked it again and the Double fault problem exist in posts 8, 9, 10, 11 and 12.
And i think you are right every where this error happens, it is a Double fault.

hamidrezakp, 2021-02-08:
This was due to an environment variable that caused the wrong image file to be build.
And the issue is not about `x86_64` crate.

closed 2021-02-08 by hamidrezakp