rust-osdev/x86_64 #232: Doesn't compile on 1.51.0 nightly

RMuskovets, 2021-02-06:
Error message:
```
error[E0557]: feature has been removed
 --> C:\Users\root\.cargo\registry\src\github.com-1285ae84e5963aae\x86_64-0.13.1\src\lib.rs:9:43
  |
9 | #![cfg_attr(feature = "const_fn", feature(const_in_array_repeat_expressions))]
  |                                           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^ feature has been removed
  |
  = note: removed due to causing promotable bugs

error: aborting due to previous error

For more information about this error, try `rustc --explain E0557`.
error: could not compile `x86_64`
```

phil-opp, 2021-02-06:
Thanks for reporting! This should be fixed already by https://github.com/rust-osdev/x86_64/pull/230. Try running `cargo update` to update to version 0.13.2.

RMuskovets, 2021-02-06:
Will try, thanks!

closed 2021-02-06 by RMuskovets