rust-osdev/x86_64 #234: Compile error: feature removed

Rdna123, 2021-02-24:
I am on Windows 10, using lasted nightly toolchain
---
error[E0557]: feature has been removed
 --> C:\Users\chknm\.cargo\registry\src\github.com-1ecc6299db9ec823\x86_64-0.12.3\src\lib.rs:9:43
  |
9 | #![cfg_attr(feature = "const_fn", feature(const_in_array_repeat_expressions))]
  |                                           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^ feature has been removed
  |
  = note: removed due to causing promotable bugs
error: could not compile `x86_64`

phil-opp, 2021-02-24:
Thanks for reporting! This was already fixed in [`v0.13.2`](https://github.com/rust-osdev/x86_64/blob/master/Changelog.md#0132--2021-02-02). There were a number of breaking changes in [`v0.13.0`](https://github.com/rust-osdev/x86_64/blob/master/Changelog.md#0130--2020-12-28) through, so the update might require some adjustments. Let me know if that works for you!

Rdna123, 2021-02-24:
I am using it in your blog-os tutorial and one of the crates depends on 0.12.3 am not sure which one though.

phil-opp, 2021-02-24:
Try using [`cargo tree`](https://doc.rust-lang.org/cargo/commands/cargo-tree.html) to find this out

Rdna123, 2021-02-25:
It was on [uart_16550](https://github.com/rust-osdev/uart_16550) I will close the issue here and open it there.

closed 2021-02-25 by Rdna123