rust-osdev/x86_64 #241: consider specifying the type arguments in the function call

Luis-Hebendanz, 2021-04-05:
Hi~,

I am not sure if this is my fault or the libraries fault but I can't get generics to work when using PageSize.

This is my function:
```rust
pub unsafe fn id_map<T: PageSize>(
    mapper: &mut (impl Mapper<T> + Translate),
    frame_allocator: &mut (impl FrameAllocator<Size4KiB> + ?Sized),
    addr: PhysAddr,
    writable: bool,
) -> Result<Page<T>, MapToError<T>> {
    let frame = PhysFrame::containing_address(addr);
    let addr = VirtAddr::new(addr.as_u64());
    // Seek for page & phys frame containing address
    let page = Page::<T>::containing_address(addr);
    use x86_64::structures::paging::PageTableFlags as Flags;
    let mut my_flags = Flags::PRESENT | Flags::NO_EXECUTE;
    my_flags.set(Flags::WRITABLE, writable);

    match mapper.translate(addr) {
        TranslateResult::NotMapped => mapper
            .identity_map(frame, my_flags, frame_allocator)?
            .flush(),
        TranslateResult::InvalidFrameAddress(_) => return Err(MapToError::FrameAllocationFailed),
        TranslateResult::Mapped { flags, .. } => {
            if flags != my_flags {
                mapper.update_flags(page, flags).unwrap().flush();
            }
        }
    };
    Ok(page)
}
```

This is how it is called:
```rust
pub unsafe fn map_and_read_phys<T: Copy>(
    mapper: &mut OffsetPageTable,
    frame_allocator: &mut impl FrameAllocator<Size4KiB>,
    addr: PhysAddr,
) -> T {
    // Map the start address
    // id_map(mapper, frame_allocator, addr, false).unwrap();
    id_map(mapper, frame_allocator, addr, false).unwrap();

    // Add type size and map if on new page
    let size = core::mem::size_of::<T>() as u64;
    id_map(mapper, frame_allocator, addr + size, false).unwrap();

    // NOTE: Can't use read_volatile because pointer is not necessarily aligned
    // like in acpi when searching for tables (as by spec)
    // core::ptr::read_volatile(addr.as_u64() as *const T)
    let ptr = addr.as_u64() as *const T;
    *ptr
}

```

and this is the error
```rust

error[E0283]: type annotations needed
   --> src/apic.rs:138:9
    |
138 |           crate::memory::id_map_nocache(
    |           -^^^^^^^^^^^^^^^^^^^^^^^^^^^^
    |           |
    |  _________cannot infer type for type parameter `T` declared on the function `id_map_nocache`
    | |
139 | |             mapper,
140 | |             frame_allocator,
141 | |             x86_64::PhysAddr::new(APIC_BASE),
142 | |             false,
143 | |         )
144 | |         .unwrap();
    | |_________________- this method call resolves to `T`
    | 
   ::: src/memory.rs:127:33
    |
127 |   pub unsafe fn id_map_nocache<T: PageSize>(
    |                                   -------- required by this bound in `id_map_nocache`
    |
    = note: cannot satisfy `_: PageSize`
help: consider specifying the type arguments in the function call
    |
138 |         crate::memory::id_map_nocache::<T, impl Mapper<T> + Translate, impl FrameAllocator<Size4KiB> + ?Sized>(
    |                                      ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^

error[E0283]: type annotations needed
  --> src/memory.rs:80:5
   |
80 |     id_map(mapper, frame_allocator, addr, false).unwrap();
   |     ^^^^^^-----------------------------------------------
   |     |
   |     this method call resolves to `T`
   |     cannot infer type for type parameter `T` declared on the function `id_map`
...
96 | pub unsafe fn id_map<T: PageSize>(
   |                         -------- required by this bound in `id_map`
   |
   = note: cannot satisfy `_: PageSize`
help: consider specifying the type arguments in the function call
   |
80 |     id_map::<T, impl Mapper<T> + Translate, impl FrameAllocator<Size4KiB> + ?Sized>(mapper, frame_allocator, addr, false).unwrap();
   |           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
```

I tried many things like calling `id_map<Size4KiB>(mapper, frame_allocator, addr, false).unwrap();` but this fails with
`explicit generic argument not allowed`.
Any idea what I'm missing here?

My rustc version:
```bash
$ rustc --version                                                                                                                                                                          101 ↵
rustc 1.51.0-nightly (a2f8f6281 2021-01-27)
```

phil-opp, 2021-04-05:
The `explicit generic argument not allowed` error occurs because it is not possible to specify generic arguments explicitly when using `impl Trait` in argument position: https://github.com/rust-lang/rust/issues/78936. As a workaround, you can assign the `id_map` result to a `let` binding with a specified type, e.g. `let x: Page<Size4KiB> = id_map(...).unwrap();`.

In general, I would probably redesign the function so that the page size is inferred from an input argument. This way, we avoid that the behavior suddenly changes if we assign the function result to a binding with a different type. In your case, it would make sense to pass in a `PhysFrame<S>` instead of the raw address like the `identity_map` function of the `x86_64` crate does: https://docs.rs/x86_64/0.13.5/x86_64/structures/paging/mapper/trait.Mapper.html#method.identity_map

Luis-Hebendanz, 2021-04-05:
I see! Man I love rust but their type system sometimes is a nightmare. 
Thank you very much! :-)

closed 2021-04-05 by Luis-Hebendanz