rust-osdev/x86_64 #244: `InterruptStackFrame`'s Pushed Registers

MarcoCicognani, 2021-04-13:
Hi!
How i can easily access the pushed registers from an `InterruptStackFrame` instance?
I need this to read/write the saved values, for example, for system calls arguments

phil-opp, 2021-04-14:
The `InterruptStackFrame` does not contain any general purpose registers. These need to be pushed separately. If you're using the `x86-interrupt` calling convention, the compiler will do this automatically for all registers that the function uses. However, there is no defined order, so you cannot access them.

So you probably need a custom assembly entry point for system calls. To make this easier, you can require in your OS that some/all registers can be overwritten on system calls so that the userspace application has to take care of saving/restoring them. This is possible because system calls are initiated by the userspace application, unlike hardware interrupts and exceptions. Then the assembly entry point can be just a thin `mov rdi, rax; call func; iretq` wrapper around an `extern "C func(rax_value: u64)"` Rust function.

josephlr, 2021-05-06:
@phil-opp is correct here. In general, it's not very ergonomic to use interrupts for system calls (as it requires dealing with the indeterminate order the registers are pushed/popped from the stack). The easier way is to just use the [`SYSCALL`/`SYSRET` instructions](https://wiki.osdev.org/SYSENTER#AMD:_SYSCALL.2FSYSRET), which avoid a lot of the pitfalls of interrupt based syscalls (hence why Linux and Windows generally prefer this method).

This crate provides support for setting up the [`Star`](https://docs.rs/x86_64/0.14.1/x86_64/registers/model_specific/struct.Star.html), [`LStar`](https://docs.rs/x86_64/0.14.1/x86_64/registers/model_specific/struct.LStar.html), and [`SFMask`](https://docs.rs/x86_64/0.14.1/x86_64/registers/model_specific/struct.SFMask.html) registers, but does not provide any wrappers around `syscall` or `sysret`. 

@MarcoCicognani, would Rust wrappers around the `syscall`/`sysret` instructions help address your issue?

MarcoCicognani, 2021-05-07:
Hi @josephlr!
Thanks for your answer, I completely agree with you!
Yes, a rust wrapper for `syscall`/`sysret` will help a lot