rust-osdev/x86_64 #252: Crate Does Not Compile on Rust 1.54 Nightly

HTG-YT, 2021-05-06:
I am currently compiling this crate with the Rust 1.54 nightly compiler, but it failed with several errors saying the use of unstable features without the feature gate.

### Meta

`rustc --version --verbose`:
```
rustc 1.54.0-nightly (bacf770f2 2021-05-05)
binary: rustc
commit-hash: bacf770f2983a52f31e3537db5f0fe1ef2eaa874
commit-date: 2021-05-05
host: x86_64-pc-windows-msvc
release: 1.54.0-nightly
LLVM version: 12.0.0
```

### Error output

```
error[E0658]: trait bounds other than `Sized` on const fn parameters are unstable
  --> <path redacted>\x86_64-0.14.0\src\structures\paging\frame.rs:18:6
   |
18 | impl<S: PageSize> PhysFrame<S> {
   |      ^
   |
   = note: see issue #57563 <https://github.com/rust-lang/rust/issues/57563> for more information
   = help: add `#![feature(const_fn_trait_bound)]` to the crate attributes to enable

error[E0658]: trait bounds other than `Sized` on const fn parameters are unstable
  --> <path redacted>\x86_64-0.14.0\src\structures\paging\page.rs:62:6
   |
62 | impl<S: PageSize> Page<S> {
   |      ^
   |
   = note: see issue #57563 <https://github.com/rust-lang/rust/issues/57563> for more information
   = help: add `#![feature(const_fn_trait_bound)]` to the crate attributes to enable

error[E0658]: trait bounds other than `Sized` on const fn parameters are unstable
   --> <path redacted>\x86_64-0.14.0\src\structures\paging\page.rs:150:6
    |
150 | impl<S: NotGiantPageSize> Page<S> {
    |      ^
    |
    = note: see issue #57563 <https://github.com/rust-lang/rust/issues/57563> for more information
    = help: add `#![feature(const_fn_trait_bound)]` to the crate attributes to enable

error: aborting due to 3 previous errors

For more information about this error, try `rustc --explain E0658`.
error: could not compile `x86_64`
```

HTG-YT, 2021-05-06:
This *looks* like a `rustc` nightly to nightly regression - as from the CI history of your crate, I find that it only started to emit the error starting from 1.53 nightly, on 27th of April.

josephlr, 2021-05-06:
Correct this is due to changes in `const_fn` Nightly features. This is a known issue and will be fixed when #250 is merged.

closed 2021-05-06 by phil-opp

phil-opp, 2021-05-06:
Should be fixed in version `0.14.1`.

PR #250 (2021-05-06): Use new `const_fn_trait_bound` feature to fix build on latest nightly
Feature gate introduced in https://github.com/rust-lang/rust/pull/84310.

Unfortunately, unknown feature gates lead to errors so that this change errors on older nightlies. For this reason, I think we should at least wait until the common `rustfmt` component is fixed on nightly again because then everyone should be able to update. See https://rust-lang.github.io/rustup-components-history/ for component availability.

Fixes #252