rust-osdev/x86_64 #262: 0.15 Tracking issue

josephlr, 2021-06-04:
This issue tracks any breaking changes we want to make for `0.15`. 

**NOTE**: All breaking changes should be made against the `next` branch, this branch should then be merged in right before releasing `0.15`. Also, remember to update `Changelog.md`.

- [x] Change `software_interrupt!()` macro to be a cosnt generic function `software_interrupt<const NUM: u8>()`. Done in #227 (and also #259 and #260)
- [x] Change the `idt::Entry` and `idt::EntryOptions` structures to move the code selector into `EntryOptions`. Also, introduce the `InterruptFn` trait which allows for abstracting over all function pointers that can be used as interrupt handlers. Done in #226 (and also in #261)
- [ ] Use [`step_trait`](https://github.com/rust-lang/rust/issues/42168) to implement `Step` for `PhysFrame` and `Page`. See #212 . This may or may not be a breaking change, depending on how we handle `PageRange`/`PageRangeInclusive`/etc....

josephlr, 2022-03-24:
Note that `const_fn_fn_ptr_basics` and `const_fn_trait_bound` were stabilized in 1.61

@phil-opp @Freax13 would we want increase our [minimum rust version](https://github.com/rust-lang/cargo/pull/9732) to be `1.61` for `v0.15.0`? Would we feel comfortable increasing the minimum rust version in a patch release? Do we have a policy on this?

phil-opp, 2022-03-24:
Ah, that's great! Increasing the minimum Rust version for `v0.15` sounds fine to me in general, as it is a semver-incompatible release with breaking changes. However, it's still 8 (?) weeks until Rust `1.61` becomes stable, so requiring it would mean that we have to postpone this release until May if we want it to keep supporting stable Rust. So I think it would be better to do this as part of a future `v0.16` release. Until then, we could use the [`rustversion`](https://github.com/dtolnay/rustversion) macro to provide the new const functionality when a new enough compiler is used.

josephlr, 2022-03-24:
Oh that crate looks amazing, it would let us get rid of our `const_fn!` macro entirely!!

If we don't need Rust 1.61 for `v0.15`, then I think we should have the MSRV be 1.56 so that we can use the [`rust-version` field](https://doc.rust-lang.org/nightly/cargo/reference/manifest.html#the-rust-version-field) in our crate.

phil-opp, 2022-03-24:
> If we don't need Rust 1.61 for `v0.15`, then I think we should have the MSRV be 1.56 so that we can use the [`rust-version` field](https://doc.rust-lang.org/nightly/cargo/reference/manifest.html#the-rust-version-field) in our crate.

Sure, sounds good to me!

josephlr, 2022-03-24:
Oh wait we use inline assembly, so the min stable version is already 1.59

josephlr, 2022-03-31:
One thing I was considering for 0.15 were changes to `VirtAddr`'s underlying representation on 64-bit platforms. If we changed `VirtAddr` to look like:
```rust
#[repr(transparent)]
pub struct VirtAddr(*const ());
```
on 64-bit platforms, we could then have the following `const fn` methods:
```rust
impl VirtAddr {
    // Safety: raw pointer has to be canonical
    #[cfg(target_pointer_width = "64")]
    pub const unsafe fn from_ptr_unsafe<T>(ptr: *const T) -> Self {
        Self(ptr as *const ())
    }

    #[cfg(target_arch = "x86_64")]
    pub const fn from_ref<T>(r: &T) -> Self {
        // Safety: references point to valid data, so are canonical.
        unsafe { Self::from_ptr_unsafe(r) }
    }
}
```

This would make it possible to construct something like a `TaskStateSegment`, `DescriptorTablePointer`, or some other structure containing a `VirtAddr` at compile time.

phil-opp, 2022-03-31:
We probably need to think about pointer provenance at some point. There is currently an [open proposal](https://github.com/rust-lang/rust/issues/95228) to make the usize to pointer conversion more strict. See the [_Rust's Unsafe Pointer Types Need An Overhaul_](https://gankra.github.io/blah/fix-rust-pointers/) post for some background information.

I'm not sure what the correct approach for our `VirtAddr` type would be. On one hand, we treat it like a pointer, e.g. in the `VirtAddr::as_ptr` method. On the other hand, the type should be allowed to cross address space bounds (e.g. kernels that keep track of the stack pointers of preempted threads), which might (?) be invalid under strict provenance rules. So maybe we need to split this type up at some point, e.g. by removing all pointer-related methods form `VirtAddr` and using pointer types in more places.

Given this issue, I don't think that it's a good idea to switch `VirtAddr` between pointer-based and u64-based depending on the platform. It would lead to more confusion and less clear semantics.

Freax13, 2022-03-31:
> Given this issue, I don't think that it's a good idea to switch `VirtAddr` between pointer-based and u64-based depending on the platform. It would lead to more confusion and less clear semantics.

I agree with that, having some methods only be const on 64-bit targets seems unintuitive. That being said I'd still be interested in potential use cases.

josephlr, 2022-04-02:
Those articles are very interesting, and I agree that changing things in this space is really complicated. It definitely seems like keeping `VirtAddr` as a `u64` is the best bet for now.

> So maybe we need to split this type up at some point, e.g. by removing all pointer-related methods form `VirtAddr` and using pointer types in more places.

This is a good idea, I'll see if I can put some examples together in a separate tracking issue.

josephlr, 2022-04-16:
@phil-opp @Freax13 is there anything else we want to get in before 0.15? All the currently pending PRs are additive, so could be done after we release 0.15

Freax13, 2022-04-17:
I can't think of anything.

phil-opp, 2022-04-18:
Me neither!

phil-opp, 2023-03-08:
What's the state of this? Should we finally prepare the 0.15 release?

Freax13, 2023-03-08:
I'll get started on rebasing next onto master.

Freax13, 2023-03-08:
> I'll get started on rebasing next onto master.

Hm, never mind, I forgot we don't do rebases. I'll get started on merging next into master instead.

Freax13, 2023-03-08:
> > I'll get started on rebasing next onto master.
> 
> Hm, never mind, I forgot we don't do rebases. I'll get started on merging next into master instead.

I'm done with that. Depending on whether on not we want to wait for some other breaking prs to also be included, I can create a pr.