rust-osdev/x86_64 #276: When will v0.14.4 be released?

toku-sa-n, 2021-07-16:
These PRs are merged into master and not breaking changes (I think):
- https://github.com/rust-osdev/x86_64/pull/258
- https://github.com/rust-osdev/x86_64/pull/267
- https://github.com/rust-osdev/x86_64/pull/269
- https://github.com/rust-osdev/x86_64/pull/271

However, v0.14.4 is not yet released. When will it be?

josephlr, 2021-07-16:
@phil-opp do you want to cut a release after #273, #274, #270 get merged?

phil-opp, 2021-07-17:
Yes, sounds good! I'm also happy to do more frequent releases for non-breaking changes in the future.

I also opened https://github.com/rust-osdev/x86_64/issues/277 to track my idea of using GitHub Actions for automatic release generation. I'm currently a bit busy with other things, but I try to look into this soon.

josephlr, 2021-07-17:
@phil-opp All the PRs I mentioned above are merged, so I think we can cut `0.14.4` anytime.

josephlr, 2021-07-19:
@phil-opp do you want me to write the 0.14.4 PR, or are you already working on it?

phil-opp, 2021-07-19:
@josephlr I probably wont have time for it until later today, so go ahead if you like.

josephlr, 2021-07-19:
> @josephlr I probably wont have time for it until later today, so go ahead if you like.

Cool, I'm almost done.

PR #283 (2021-07-19): Release version to 0.14.4
Lets see if the CI works!

We will probably want to merge this instead of squashing it (to preserve git history)

Close #276
closed 2021-07-19 by josephlr