rust-osdev/x86_64 #284: Abstractions over debug registers (DR0-DR7)

mkroening, 2021-07-19:
Working on the GDB support for [uhyve](https://github.com/hermitcore/uhyve), I wrote abstractions for the debug registers (DR0-DR7) to be able to easily configure hardware breakpoints. See a work-in-progress and undocumented version [here](https://github.com/mkroening/uhyve/blob/a7d01d38fb99eeb56f07156bcb091a38f717135f/src/arch/x86_64/registers/debug.rs).

Would such abstractions be in scope of this project?

josephlr, 2021-07-19:
I would think so, it seems like configuring the debug registers could work similar to our configuration of the control registers, see [`x86_64::registers::control`](https://docs.rs/x86_64/0.14.4/x86_64/registers/control/index.html).

I think if we added a `x86_64::registers::debug` module with similar abstractions.
  - We could have a way to read/write the `VirtAddr` for `Dr0`/`Dr1`/`Dr2`/`Dr3`
    - similar to what we currently do for `Cr2`
    - These should probably have a common type or trait
  - We would have `Dr6`/`D6Flags` and `Dr7`/`Dr7Flags`
    - similar to what we do for `Cr0` and `Cr4`
    - `Dr6` might make more sense to just have `read` and `clear` methods
    - `Dr7` should have `read` and `write` methods

josephlr, 2021-07-19:
In addition to the AMD/Intel manuals, it might also be useful to look at [how Linux does things](https://github.com/torvalds/linux/blob/69f737ed3a1c0e7619565bb4c01dff5ec8991d85/arch/x86/include/uapi/asm/debugreg.h)

mkroening, 2021-07-21:
Thanks for the quick reply! I'll prepare a PR soon. :+1:

Regarding `Dr7Flags`: There are fields (`R/W` and `LEN`), which are two bits long. Using `bitflags!` for those wouldn't work nicely, would it?