rust-osdev/x86_64 #294: Failed Tests on `Install QEMU (Windows)` stage

ncatelli, 2021-08-14:
Tests are consistently failing on the `Install QEMU (Windows)` stage due to a `Can't shim 'qemu-system-moxie.exe': File doesn't exist.` error. I found a similarly described issue on the scoop project at https://github.com/lukesampson/scoop/issues/2861, though the fix did not seem immediately clear.

ncatelli, 2021-08-14:
Looks like phil-opp beat me to a previous issue https://github.com/lukesampson/scoop/issues/4183

ncatelli, 2021-08-14:
Fix in https://github.com/ScoopInstaller/Main/pull/2553

josephlr, 2021-08-16:
I think we should directly download the installer from https://qemu.weilnetz.de/w64/2021/

That's all scoop is doing anyway, so it seems weird to download an installer that just essentially downloads another installer.

@phil-opp is there any reason we shouldn't just download qemu directly?

ncatelli, 2021-08-16:
I assume it's for the easy unattended install. But if the preference is to do away with scoop, I can sort out a PR to get that working.

phil-opp, 2021-08-17:
I think I originally used scoop to download multiple dependencies and because it was easier to set up. However, it already took down our CI pipeline a few times, so maybe installing qemu directly would be better. This would also give us the option to fix any issues ourselves instead of waiting for fixes in scoop.

phil-opp, 2021-08-17:
This of course all depends on how much effort the manual install would be. If it requires more than ~2-4 commands, sticking to scoop is probably the better option.

josephlr, 2021-08-17:
So in #287 I changed the CI to just do `choco install qemu`, which works and properly installs QEMU. 

However, now the tests timeout. I'm guessing it has something to do with the arguments passed to QEMU being different on Windows vs Linux. For now, I've just changed things to allow the Windows tests to fail to unblock things.

@phil-opp does it even make sense to run the QEMU tests on non-Linux platforms? Running the same binary on QEMU on multiple platforms seems unnecessary. At that point it seems like we're testing QEMU, not the `x86_64` crate.

My preference would be to only run the QEMU integration tests on Linux. We would still run the `cargo build`/`cargo test`/etc... commands on Linux, macOS, and Windows.

phil-opp, 2021-08-18:
> So in #287 I changed the CI to just do `choco install qemu`, which works and properly installs QEMU.

Good solution, thanks! I didn't know that GitHub Action runners now have choco installed by default.

> However, now the tests timeout. I'm guessing it has something to do with the arguments passed to QEMU being different on Windows vs Linux. For now, I've just changed things to allow the Windows tests to fail to unblock things.

Hmm, the tests still worked [a week ago](https://github.com/rust-osdev/x86_64/runs/3297197980?check_suite_focus=true), so it would be good to get them working again. The only reason that the Windows tests are not marked as required is that the install step failed a few times. The tests themselves normally worked without problems.

> does it even make sense to run the QEMU tests on non-Linux platforms? Running the same binary on QEMU on multiple platforms seems unnecessary. At that point it seems like we're testing QEMU, not the `x86_64` crate.

I think it makes sense to run it in QEMU on all major platforms because our code should not cause problems on any platform (people are using this crate in QEMU on Windows). Another advantage of the current approach is that we test our code against the common QEMU versions for each OS, which can be very different. For example, the latest CI job of #287 (which timed out on Windows) uses QEMU 4.2.1 on Linux, v6.0.0 on macOS, and v6.0.93 on Windows. So the version used on Windows is the newest, so I think it's more likely that the issue is caused by some breaking change in QEMU and not by platform differences.

phil-opp, 2021-08-18:
>  So the version used on Windows is the newest, so I think it's more likely that the issue is caused by some breaking change in QEMU and not by platform differences.

Given that the Windows seems to be a release candidate version, so it might be a regression (instead fo a breaking change). Maybe it works if we force the version to v6.0.0. I just opened https://github.com/rust-osdev/x86_64/pull/297 to try that.

phil-opp, 2021-08-18:
Seems like using version 6.0.0 on Windows fixes the timeout problems: https://github.com/rust-osdev/x86_64/runs/3358253309?check_suite_focus=true (https://github.com/rust-osdev/x86_64/pull/297)

josephlr, 2021-08-18:
@phil-opp awesome to hear!! Thanks for looking into this.

josephlr, 2021-08-19:
With #297 merged this is fixed.

closed 2021-08-19 by josephlr