rust-osdev/x86_64 #295: How to create an instance of `MapperFlush`?

toku-sa-n, 2021-08-15:
I was trying to implement `Mapper` for my process struct and noticed that `MapperFlush::new` is a private method. Is there a way to create an instance of it?

josephlr, 2021-08-20:
From the [`MapperFlushAll` docs](https://docs.rs/x86_64/0.14.4/x86_64/structures/paging/mapper/struct.MapperFlushAll.html):
> This type is returned from a function that made the change to ensure that the TLB flush is not forgotten.

Basically, you are not supposed to be able to create this type. It is returned from functions like: [`Mapper::set_flags_p4_entry`](https://docs.rs/x86_64/0.14.4/x86_64/structures/paging/mapper/trait.Mapper.html#tymethod.set_flags_p4_entry) forcing the user to explicitly do something with the `MapperFlush` object.

See [this example](https://docs.rs/x86_64/0.14.4/x86_64/structures/paging/mapper/trait.Mapper.html#examples) for how the type is used.

toku-sa-n, 2021-08-20:
Then, is it not supposed to implement `Mapper` manually? Implementing it needs to create an instance of `MapperFlush` and `MapperFlushAll`.

josephlr, 2021-08-20:
@toku-sa-n whoops my bad, I misread your initial comment. Yes I believe external implementations of mapper should be supported, so we do need to add the constructors for this to be possible.

Good catch.

PR #296 (2021-08-29): feat(mapper): expose `MapperFlush(All)?::new`
It looks like there is no way to create an instance of these types.

Fixes: #295
commit 278e1fee (2021-08-29): feat(mapper): expose `MapperFlush(All)?::new`

It looks like there is no way to create an instance of these types.

Fixes: #295
closed 2021-08-29 by josephlr