rust-osdev/x86_64 #307: `cargo doc` is broken when `default-featues = false`

npmccallum, 2021-09-16:
This is particularly nasty because it also breaks `cargo doc` on all downstream uses of this crate with `default-features = false`.

npmccallum, 2021-09-16:
```
$ cargo doc --no-default-features
   Compiling x86_64 v0.14.4 (/home/npmccallum/enarx/x86_64)
    Checking volatile v0.4.4
    Checking bit_field v0.9.0
 Documenting bit_field v0.9.0
 Documenting volatile v0.4.4
 Documenting bitflags v1.3.2
    Checking bitflags v1.3.2
 Documenting x86_64 v0.14.4 (/home/npmccallum/enarx/x86_64)
error[E0433]: failed to resolve: could not find `instructions` in the crate root
 --> src/registers/model_specific.rs:5:5
  |
5 |     instructions::segmentation::{Segment64, FS, GS},
  |     ^^^^^^^^^^^^ could not find `instructions` in the crate root

error[E0433]: failed to resolve: could not find `instructions` in the crate root
 --> src/structures/gdt.rs:4:12
  |
4 | use crate::instructions::segmentation::{Segment, CS, SS};
  |            ^^^^^^^^^^^^ could not find `instructions` in the crate root

thread 'rustc' panicked at 'internal error: entered unreachable code', compiler/rustc_resolve/src/lib.rs:3341:67
note: run with `RUST_BACKTRACE=1` environment variable to display a backtrace

error: internal compiler error: unexpected panic

error: Unrecognized option: 'crate-version'

For more information about this error, try `rustc --explain E0433`.
error: could not document `x86_64`
```

phil-opp, 2021-09-17:
Thanks for reporting! I agree that this is unfortunate and should be fixed as soon as possible. I'm not sure if #308 is a good solution to this (see my comment there), but I'm currently working on a different solution.

phil-opp, 2021-09-17:
> I'm currently working on a different solution.

I opened a PR in https://github.com/rust-osdev/x86_64/pull/309.

commit a280bbe1 (2021-09-17): Move segment types into a new `registers::segmentation` module

The segment types themselves do not require the `instructions` feature, so it makes sense to move them to the top-level `registers` module. The `instructions::segmentation` module still re-exports all the types, so this change should not break any code.

Fixes #307
PR #309 (2021-09-20): Move segment types into a new `registers::segmentation` module
The segment types themselves and the related traits do not require the `instructions` feature, so I think that it makes sense to move them to the top-level `registers` module. The `instructions::segmentation` module still re-exports all the types, so this change should not break any code. I also moved the `SegmentSelector` type from the `gdt` module to the new module since it's more related to the segment registers than to the GDT.

This should fix `cargo doc` errors reported in #307 since the segment types are now available even w [... cut]
closed 2021-09-20 by josephlr