rust-osdev/x86_64 #325: Add a method to get an immutable reference to the level 4 table

pjht, 2021-11-16:
The current mappers provide a method to get a mutable reference to the level 4 table. I have found that a mutable reference is (at least for my usecases), unnecessary and causes borrow checker issues. I would like to add a method to get an immutable reference instead, maybe called `level_4_immut`.

Freax13, 2021-11-16:
That seems reasonable. I think the most idiomatic way to deal with this would be to make `level_4_table` return an immutable reference and to add `level_4_table_mut` which returns a mutable reference, however that would be a breaking change.

We could either introduce that change for the next breaking release, add `level_4_table_immut` right now or do both, by adding `level_4_table_immut` now and later renaming `level_4_table_immut` and `level_4_table` to `level_4_table` and `level_4_table_mut` respectivly in the next breaking release.

Freax13, 2021-12-02:
I went ahead and created #327. If merged this will be part of the next breaking release.

closed 2021-12-05 by Freax13