rust-osdev/x86_64 #332: Call to VirtAddr::new() causes panic when CR2 does not contain a canonical address.

RevolutionXenon, 2021-12-29:
https://docs.rs/x86_64/latest/src/x86_64/registers/control.rs.html#267

When CR2 contains a malformed address, this panic occurs:
"panicked at 'address passed to VirtAddr::new must not contain any data in bits 48 to 64"

CR2 can obviously contain a non-canonical address after a page fault. This panic should not occur. I'm not sure what the best fix would be, but I ran into this because I tried to print CR2 in my panic handler and it just starts looping when you do that.

Freax13, 2021-12-29:
Good catch we should definitely not panic in that situation. We could change `Cr2::read` to return `Result<VirtAddr, VirtAddrNotValid>` and add a new method `Cr2::read_raw` that returns `u64`. 

Changing `Cr2::read` would be a breaking change, but adding `Cr2::read_base` could be done right now.

josephlr, 2022-01-18:
Should this stay open to track the (breaking) change to `Cr2::read`?

josephlr, 2022-01-18:
Whoops I just saw #335 I guess that PR (when merged) should close this issue.

toku-sa-n, 2022-02-07:
I think this issue can be closed.

PR #334 (2022-01-17): add `Cr2::read_raw`
This pr adds `Cr2::read_raw`.

Closes #332
PR #335 (2022-02-06): make `Cr2::read` return a result
This pr adds `Cr2::read_raw` and makes `Cr2::read` return a result.

Closes #332
closed 2022-02-07 by Freax13