rust-osdev/x86_64 #372: Add CPUID instruction

adsnaider, 2022-04-08:
Not sure why this isn't already part of the API, it seems like a very natural instruction to have. Am I missing something?

josephlr, 2022-04-08:
Nope, it's mostly just figuring out the correct abstraction around the instruction. CPUID does a lot of things, and returns a bunch of different types of data. We will likely need a combination of a "raw" and "non-raw" CPUID function.

adsnaider, 2022-04-08:
Tbh, I was just thinking of a low level function like this one

```rust
use core::arch::asm;

/// Runs the CPUID instruction with the provided `eax` register.
///
/// # Returns
///
/// A 4-tuple (eax, ebx, ecx, edx). It's up to the user to interpret these values based on the
/// input.
pub fn cpuid(mut eax: u32) -> (u32, u32, u32, u32) {
    let ebx: u32;
    let ecx: u32;
    let edx: u32;
    unsafe {
        asm!(
            "push rbx",
            "cpuid",
            "mov esi, ebx",
            "pop rbx",
             out("esi") ebx,
             inout("eax") eax,
             out("ecx") ecx,
             out("edx") edx
        );
    }
    (eax, ebx, ecx, edx)
}
```

Even this I think would be useful as it's non-trivial given that LLVM uses rbx.

josephlr, 2022-04-08:
So they're already are some functions related to CPUID in the standard library `core::arch::x86_64` module:
  - [`__cpuid_count`](https://doc.rust-lang.org/core/arch/x86_64/fn.__cpuid_count.html) (unsafe)
  - [`__cpuid`](https://doc.rust-lang.org/core/arch/x86_64/fn.__cpuid.html) (unsafe)
  - [`has_cpuid`](https://doc.rust-lang.org/core/arch/x86_64/fn.has_cpuid.html) (safe, unstable)
  - [`__get_cpuid_max`](https://doc.rust-lang.org/core/arch/x86_64/fn.__get_cpuid_max.html) (unsafe)

I would want to make sure we're doing something more then just reimplementing them. That could just be providing a safe and/or stable API.

Regardless, we should reuse the [`CpuidResult`](https://doc.rust-lang.org/core/arch/x86/struct.CpuidResult.html) type.

Freax13, 2022-04-08:
Just wanted to point out that there's already a very good crate that implements abstractions over cpuid: [`raw-cpuid`](https://crates.io/crates/raw-cpuid).

josephlr, 2022-04-08:
@Freax13 That's a really good point. Should we just add a note to the README mentioning that crate? It could be in a "other helper crates" section.

adsnaider, 2022-04-08:
@Freax13 Thank you for the info. I agree that having that linked in the README would be sufficient in my opinion.

PR #382 (2022-04-20): Update comment and docs
First commit adds a Safety comment to `FrameAllocator`, fixing a clippy warning.

Second commit adds links to other useful OS crates. Fixes #372
commit 43afc598 (2022-04-20): Add links to other useful OS crates

Fixes #372
closed 2022-04-20 by josephlr