rust-osdev/x86_64 #390: Mapper always requires FrameAllocator<Size4Kib>

tsatke, 2022-08-26:
The trait `Mapper<S: PageSize>` has a `FrameAllocator<Size4Kib>` hard-coded in these locations.

https://github.com/rust-osdev/x86_64/blob/master/src/structures/paging/mapper/mod.rs#L188
https://github.com/rust-osdev/x86_64/blob/master/src/structures/paging/mapper/mod.rs#L280
https://github.com/rust-osdev/x86_64/blob/master/src/structures/paging/mapper/mod.rs#L368

I think that should rather be a `FrameAllocator<S>`, correct me if I'm wrong.

Freax13, 2022-08-26:
> The trait `Mapper<S: PageSize>` has a `FrameAllocator<Size4Kib>` hard-coded in these locations.

The `FrameAllocator<Size4Kib>` is required because mapping a page might require creating new page tables and page tables happen to have a size of 4KiB. 

> I think that should rather be a `FrameAllocator<S>`, correct me if I'm wrong.

The methods you listed already take a frame of the appropriate size as a parameter, so there's need to allocate a new one which is why the `FrameAllocator<S>` bound is not needed.

tsatke, 2022-08-26:
Ah, got it, thanks!

closed 2022-08-26 by tsatke