rust-osdev/x86_64 #391: OffsetPageTable does not use offset for creating a new mapping

0x6D70, 2022-09-19:
When a ``OffsetPageTable`` is created with an offset, the offset is not used for accessing the page table when mapping a page.

# Example

For example:
```rust
let mut pt = OffsetPageTable::new(page_table, VirtAddr::new(0xffff800000000000));
pt.map_to(...); // here the page tables are accessed without an offset even tough an OffsetPageTable with an offset is used
```

# Problem

As far as I can see the problem occurs in ``map_to_4kib`` (Note: the same function for other sizes are also affected) calls ``self.page_table_walker.create_next_table`` with the page table entry. The ``create_next_table`` function accesses the page for checking if the entry is unused.

The problem is that nowhere the conversion function ``frame_to_pointer`` from the ``PageTableFrameMapping`` is used.

0x6D70, 2022-09-19:
No, ``page_table`` is the physical address of the level 4 page table

0x6D70, 2022-09-19:
[image]
When  adding the offset, the ``map_to`` function panics. I don't know if this panic has anything to do with the other problem.

0x6D70, 2022-09-20:
Oh, I forgot to set the Present bit for the mapping. Is there a reason why the present bit is not automatically set?

@Freax13 as you said, the documentation should mention that the offset should be added to the level 4 table

Freax13, 2022-09-20:
> Oh, I forgot to set the Present bit for the mapping. Is there a reason why the present bit is not automatically set?

I suppose it's possible to want to map a page without the present bit, but in that case we should still set the present flag on the page tables. This still won't let you access the mapped page, but prevents the crash.

0x6D70, 2022-09-20:
👍 , thanks for your help

closed 2022-09-20 by 0x6D70