rust-osdev/x86_64 #392: [Problem] Add software interrupt.

AmberCrafter, 2022-09-23:
I would like to implment the software interrupt on idt[0x80] ( likes linux's syscall ), but I've run into a calling convention problem:
`eax` will be overwrited when `extern "x86-interrupt" fn foo {}` is executed.
Is this the feature of the `x86-interrupt`?  Or am I missing anything?
And where can I found the doucuments about `x86-interrupt`?

[image]
[image]
[image]

AmberCrafter, 2022-09-23:
[RFC](https://github.com/rust-lang/rfcs/pull/3246#issuecomment-1088935508) mention `x86-interrupt` was clobbered `rax`.
Is that mean we need to use other register to pass the syscall's number? More over, should we need to consider other registers are callee-save registers or define them by myself?

Freax13, 2022-09-23:
AFAICT there are no guarantees about the contents of registers once the function has been entered. If you want to inspect/modify registers you could use the [`#[naked]` attribute on a function](https://rust-lang.github.io/rfcs/1201-naked-fns.html) and implement the calling convention yourself.

AmberCrafter, 2022-09-26:
@Freax13 Thanks for your suggestion. 
In these days, I trace the asm code of different kind x86-inerrupt function, and found the interesting behavior:

If we use the InterruptStackFrame, the asm code like below, which will overwrite the `rax`
[image]

however, if we unuse the InterruptStackFrame, it looks like this, which will not overwrite the `rax`.
[image]

On the other hand, @xfoxfu ([repo.](https://github.com/xfoxfu/rust-xos/blob/main/kernel/src/interrupts/handlers.rs#L21)) provides a good solution to do the type wrapping (extern "C" -> extern "x86-interrupt") which has already perplexed me for a long time.

Freax13, 2022-09-26:
> however, if we unuse the InterruptStackFrame, it looks like this, which will not overwrite the `rax`. [image]

There's no guarantee that this will always be the case. This might work for now, but can (and most certainly will) break at any time in the future.

> On the other hand, @xfoxfu ([repo.](https://github.com/xfoxfu/rust-xos/blob/main/kernel/src/interrupts/handlers.rs#L21)) provides a good solution to do the type wrapping (extern "C" -> extern "x86-interrupt") which has already perplexed me for a long time.

The thing that makes this work is their [`wrap!` macro](https://github.com/xfoxfu/rust-xos/blob/2821bcf1b61bfa0f8f7d57e5982ebbc6644bba1a/kernel/src/interrupts/handlers.rs#L112-L161) which generates a naked function that pushes the registers onto the stack and prepares the registers for calling a function with the C abi before calling the wrapped function.

I think they could use the `x86-interrupt` calling convention for their generated function and then they wouldn't have to [transmute the function](https://github.com/xfoxfu/rust-xos/blob/2821bcf1b61bfa0f8f7d57e5982ebbc6644bba1a/kernel/src/interrupts/handlers.rs#L23) when setting the entry.

closed 2022-09-26 by AmberCrafter