rust-osdev/x86_64 #393: process didn't exit successfully: - rustc --crate-name lock_api --edition=2018

FurkanGozukara, 2022-10-19:
Hello. I am trying to make work this project

It requires specific version of transformers

https://github.com/xashru/punctuation-restoration

I have tried various version of rustc but still getting this error

Which specific version of rust i need?

Python 3.10.8 (tags/v3.10.8:aaaf517, Oct 11 2022, 16:50:30) [MSC v.1933 64 bit (AMD64)] on win32

rustc 1.41.0 (5e1a79984 2020-01-27)

```
 Compiling textwrap v0.11.0
           Running `rustc --crate-name textwrap C:\Users\King\.cargo\registry\src\github.com-1ecc6299db9ec823\textwrap-0.11.0\src\lib.rs --error-format=json --json=diagnostic-rendered-ansi,artifacts,future-incompat --crate-type lib --emit=dep-info,metadata,link -C opt-level=3 -C embed-bitcode=no -C metadata=65eec7bbb3cdf0f8 -C extra-filename=-65eec7bbb3cdf0f8 --out-dir C:\Users\King\AppData\Local\Temp\pip-install-ablkymhr\tokenizers_201770b635934fc0aaf386ed5bdd381e\target\release\deps -L dependency=C:\Users\King\AppData\Local\Temp\pip-install-ablkymhr\tokenizers_201770b635934fc0aaf386ed5bdd381e\target\release\deps --extern unicode_width=C:\Users\King\AppData\Local\Temp\pip-install-ablkymhr\tokenizers_201770b635934fc0aaf386ed5bdd381e\target\release\deps\libunicode_width-5f855d5352d382a0.rmeta --cap-lints allow`
         Compiling lock_api v0.3.4
         Compiling number_prefix v0.3.0
         Compiling vec_map v0.8.2
           Running `rustc --crate-name lock_api --edition=2018 C:\Users\King\.cargo\registry\src\github.com-1ecc6299db9ec823\lock_api-0.3.4\src\lib.rs --error-format=json --json=diagnostic-rendered-ansi,artifacts,future-incompat --crate-type lib --emit=dep-info,metadata,link -C opt-level=3 -C embed-bitcode=no --cfg "feature=\"nightly\"" -C metadata=2fe17ee22c58d967 -C extra-filename=-2fe17ee22c58d967 --out-dir C:\Users\King\AppData\Local\Temp\pip-install-ablkymhr\tokenizers_201770b635934fc0aaf386ed5bdd381e\target\release\deps -L dependency=C:\Users\King\AppData\Local\Temp\pip-install-ablkymhr\tokenizers_201770b635934fc0aaf386ed5bdd381e\target\release\deps --extern scopeguard=C:\Users\King\AppData\Local\Temp\pip-install-ablkymhr\tokenizers_201770b635934fc0aaf386ed5bdd381e\target\release\deps\libscopeguard-38978501dffc7e67.rmeta --cap-lints allow`
           Running `rustc --crate-name number_prefix C:\Users\King\.cargo\registry\src\github.com-1ecc6299db9ec823\number_prefix-0.3.0\src\lib.rs --error-format=json --json=diagnostic-rendered-ansi,artifacts,future-incompat --crate-type lib --emit=dep-info,metadata,link -C opt-level=3 -C embed-bitcode=no --cfg "feature=\"default\"" --cfg "feature=\"std\"" -C metadata=18e86f8ac15d36d7 -C extra-filename=-18e86f8ac15d36d7 --out-dir C:\Users\King\AppData\Local\Temp\pip-install-ablkymhr\tokenizers_201770b635934fc0aaf386ed5bdd381e\target\release\deps -L dependency=C:\Users\King\AppData\Local\Temp\pip-install-ablkymhr\tokenizers_201770b635934fc0aaf386ed5bdd381e\target\release\deps --cap-lints allow`
         Compiling either v1.8.0
           Running `rustc --crate-name vec_map C:\Users\King\.cargo\registry\src\github.com-1ecc6299db9ec823\vec_map-0.8.2\src\lib.rs --error-format=json --json=diagnostic-rendered-ansi,artifacts,future-incompat --crate-type lib --emit=dep-info,metadata,link -C opt-level=3 -C embed-bitcode=no -C metadata=c1dc532dfe4f0140 -C extra-filename=-c1dc532dfe4f0140 --out-dir C:\Users\King\AppData\Local\Temp\pip-install-ablkymhr\tokenizers_201770b635934fc0aaf386ed5bdd381e\target\release\deps -L dependency=C:\Users\King\AppData\Local\Temp\pip-install-ablkymhr\tokenizers_201770b635934fc0aaf386ed5bdd381e\target\release\deps --cap-lints allow`
           Running `rustc --crate-name either --edition=2018 C:\Users\King\.cargo\registry\src\github.com-1ecc6299db9ec823\either-1.8.0\src\lib.rs --error-format=json --json=diagnostic-rendered-ansi,artifacts,future-incompat --crate-type lib --emit=dep-info,metadata,link -C opt-level=3 -C embed-bitcode=no -C metadata=367d734b841a6246 -C extra-filename=-367d734b841a6246 --out-dir C:\Users\King\AppData\Local\Temp\pip-install-ablkymhr\tokenizers_201770b635934fc0aaf386ed5bdd381e\target\release\deps -L dependency=C:\Users\King\AppData\Local\Temp\pip-install-ablkymhr\tokenizers_201770b635934fc0aaf386ed5bdd381e\target\release\deps --cap-lints allow`
      error[E0557]: feature has been removed
        --> C:\Users\King\.cargo\registry\src\github.com-1ecc6299db9ec823\lock_api-0.3.4\src\lib.rs:91:42
         |
      91 | #![cfg_attr(feature = "nightly", feature(const_fn))]
         |                                          ^^^^^^^^ feature has been removed
         |
         = note: split into finer-grained feature gates

      For more information about this error, try `rustc --explain E0557`.
      error: could not compile `lock_api` due to previous error

      Caused by:
        process didn't exit successfully: `rustc --crate-name lock_api --edition=2018 C:\Users\King\.cargo\registry\src\github.com-1ecc6299db9ec823\lock_api-0.3.4\src\lib.rs --error-format=json --json=diagnostic-rendered-ansi,artifacts,future-incompat --crate-type lib --emit=dep-info,metadata,link -C opt-level=3 -C embed-bitcode=no --cfg "feature=\"nightly\"" -C metadata=2fe17ee22c58d967 -C extra-filename=-2fe17ee22c58d967 --out-dir C:\Users\King\AppData\Local\Temp\pip-install-ablkymhr\tokenizers_201770b635934fc0aaf386ed5bdd381e\target\release\deps -L dependency=C:\Users\King\AppData\Local\Temp\pip-install-ablkymhr\tokenizers_201770b635934fc0aaf386ed5bdd381e\target\release\deps --extern scopeguard=C:\Users\King\AppData\Local\Temp\pip-install-ablkymhr\tokenizers_201770b635934fc0aaf386ed5bdd381e\target\release\deps\libscopeguard-38978501dffc7e67.rmeta --cap-lints allow` (exit code: 1)
      warning: build failed, waiting for other jobs to finish...
      error: `cargo rustc --lib --message-format=json-render-diagnostics --manifest-path Cargo.toml --release -v --features pyo3/extension-module -- --crate-type cdylib` failed with code 101
      [end of output]

  note: This error originates from a subprocess, and is likely not a problem with pip.
  ERROR: Failed building wheel for tokenizers
Failed to build tokenizers
ERROR: Could not build wheels for tokenizers, which is required to install pyproject.toml-based projects
```

Freax13, 2022-10-19:
This project is in no way associated with the rust compiler and it's also not the dependency that failed to compile. This issue does not belong here and I'm not quite sure why you thought it did. I saw that you already opened multiple issues in more related projects (https://github.com/huggingface/tokenizers/issues/1088, https://github.com/xashru/punctuation-restoration/issues/13, https://github.com/Amanieu/parking_lot/issues/360), so I'm going to close this.

closed 2022-10-19 by Freax13

FurkanGozukara, 2022-10-19:
@Freax13  i saw very similar issue on this repo.

yes I asked multiple places but 0 useful answer so far

Freax13, 2022-10-19:
> @Freax13 i saw very similar issue on this repo.

Ah fair enough, we've had some nightly breakages affecting this project in the past, but we don't support outdated versions on outdated compilers, so we usually just fix it and tell people to use the latest nightly. If I understood correctly, you can't do that, so that doesn't apply and I can't help you.