rust-osdev/x86_64 #395: [Question] Why at most 8 entries for the GDT?

TornaxO7, 2022-11-11:
May I ask why there can be only at most 8 entries in the GDT? According to the intel manual the GDT should be able to have at most 8.192 entries:
[image]

josephlr, 2022-11-12:
The basic answer is we had to pick _a_ size, and 8 was a reasonable value that worked for most use-cases.

In theory, with const-generics now stable, we could add a parameter to make the maximum size user-configurable. Something like:
```rust
#[derive(Debug, Clone)]
pub struct GlobalDescriptorTable<const MAX: usize = 8> {
    table: [u64; MAX],
    len: usize,
}

impl<const MAX: usize> GlobalDescriptorTable<MAX> {
   #[inline]
    pub const fn new() -> Self {
        assert!(MAX >= 1);
        Self {
            table: [0; MAX],
            len: 1,
        }
    }
    ...
}
```

The main downside is that while this _looks_ like it would be a non-breaking change, it is actually a very subtle breaking change:
```rust
pub fn this_still_works() -> GlobalDescriptorTable {
    let x = GlobalDescriptorTable::new();
    x
}
```
```rust
pub fn this_stops_working() {
    let x = GlobalDescriptorTable::new();
    ...
}
```

TornaxO7, 2022-11-12:
Hm... ok. Thank you for your response!

closed 2022-11-12 by TornaxO7

phil-opp, 2024-01-14:
Update: We merged a PR to use const generics for the GDT in #360, apparently even before this issue was opened. The reason that this was not available on crates.io yet is that the PR targeted the v0.15 release, which is almost ready now (see #446).