rust-osdev/x86_64 #396: `SelectorErrorCode::index()` parses the wrong bits

kennystrawnmusic, 2022-12-15:
According to a lengthy troubleshooting discussion on [StackOverflow](https://stackoverflow.com/questions/74807148/why-would-a-np-fault-expecting-an-idt-entry-at-index-302-even-be-possible/74811950#74811950) attempting to figure out why my interrupt handler was giving weird selector indices:

> Interesting and almost what I expected but goes against how the selector error code is documented. See just before this 100: v=97 e=0000 i=0 cpl=0 IP=0008:0000008000021b3c pc=0000008000021b3c SP=0010:fffff00007ffe8f0 env->regs[R_EAX]=000001807eb4a3e0. v=97 (that is actually 0x97). It almost appears (despite how it is documented) that in 64-bit code the selector error code has to be shifted right 4 bits rather than 3 bits (as it is in 32 bit code). It makes sense since since 64-bit entries are 16 bytes. That would yield index 0x97 (decimal 151) for the interrupt number.

Sure enough, following this advice [finally fixed the dreaded, elusive Issue #10](https://github.com/kennystrawnmusic/cryptos/commit/391a7c3e1d9dee8366b5ccf57572c7e44ea6a373) in my kernel. This can only mean one thing: a misaligned parse of the selector error code's bits by the `x86_64` crate, since I use `SelectorErrorCode::new(code)` to debug this.

mpetch, 2022-12-16:
I'm going to investigate this issue further. Between the AMD manual and the Intel Manuals they do suggest bits 3:16 are the index and that it doesn't differ for 32-bit or 64-bit modes. I'm going to run some tests on real hardware. It may well be that QEMU is at fault encoding the index in 64-bit code incorrectly.

phil-opp, 2022-12-16:
@mpetch  Thanks a lot!

mpetch, 2022-12-16:
I have to write a bit of debug code to output values on real hardware. But I can already see that BOCHS and QEMU differ! BOCHS puts a selector error code on the stack that in line with what the documentation suggests. QEMU doesn't. Once I get a test program together and run it on real hardware I expect BOCHS and the real hardware will agree. My hunch is the bug exists in QEMU now.

This isn't an issue with GDT selectors since those descriptors are 8 bytes in size (except for the TSS which takes to index entries) but all of them are still scaled by 8.

mpetch, 2022-12-16:
From my testing I am concluding QEMU software emulation has a bug.

I created a disk image that I could run on BOCHS, QEMU, and real hardware. It simulates your problem. I get the correct selector index on real hardware, BOCHS, and QEMU if using the `-enable-kvm` option. If I run QEMU without that option I get the wrong selector encoded. This suggests QEMU software emulation has a bug and improperly encodes the selector using 8 byte scaling rather than 16 byte for IDT index entries in long mode.

kennystrawnmusic, 2022-12-16:
Since we know that this is indeed a QEMU issue and not an issue with this crate after all, closing this and closing the associated pull request.

closed 2022-12-16 by kennystrawnmusic

phil-opp, 2022-12-17:
> I get the correct selector index on real hardware, BOCHS, and QEMU if using the `-enable-kvm` option. If I run QEMU without that option I get the wrong selector encoded. This suggests QEMU software [...]

Yeah, it clearly sounds like there is a bug in QEMU. I guess we should open a bug at <https://gitlab.com/qemu-project/qemu/-/issues> about it?

mpetch, 2022-12-17:
I am in the process of doing that as we speak. I put my [test code](https://github.com/mpetch/SelectorErrorCodeTest) up on GitHub with the intention of  linking it to a QEMU bug report.