rust-osdev/x86_64 #405: const fn gdt creation

tepperson2, 2023-02-28:
I would like the ability to create a GlobalDescriptorTable with a const fn. I am writing a kernel and need to ensure that no 64-bit code runs before I jump to long mode because the kernel is started in 32-bit mode. I want to define the gdt structure with rust.

tepperson2, 2023-02-28:
I just noticed the "const_fn" feature, but I would like to do this on stable rust.

phil-opp, 2023-03-01:
The `const_fn` feature is using the compiler's unstable `const_mut_refs` feature, which is hopefully stabilized soon. See https://github.com/rust-lang/rust/issues/57349 for the current state. Unfortunately, we don't have any other way to create a GDT in a const environment at the moment. If you have an idea how we could implement it, please let us know!

tepperson2, 2023-03-03:
I added a pull request https://github.com/rust-osdev/x86_64/pull/406

josephlr, 2023-03-03:
Note that you can also use [`GlobalDescriptorTable::from_raw_slice`](https://docs.rs/x86_64/latest/x86_64/structures/gdt/struct.GlobalDescriptorTable.html#method.from_raw_slice) if you want something that works on the current version of stable Rust.