rust-osdev/x86_64 #434: Enable OpenSSF Scorecard Action

joycebrum, 2023-09-06:
Hi again, I'd like to suggest the adoption of the [OpenSSF Scorecard Action](https://github.com/ossf/scorecard-action). It is a scanner that identifies which practices can be adopted to increase the project's supply-chain security posture. It also monitors the project if any change affected any of practices.

It is developed by the [open source security foundation](https://openssf.org/) in order to mitigate the known supply-chain attack vectors.

The findings can be seen at the [security dashboard](https://github.com/rust-osdev/x86_64/security). Optionally, a badge with the project's score can be added to the readme (since x86_64 score is really great, I think it would be good to disclose it to users). The current project score is 7.5, which only 0.2% of over 1 million open source projects were able to achieve. 

Let me know if you are interested in such tool and I can help on configuring it through a PR.

Thanks!

Freax13, 2023-09-06:
Personal opinion ahead, other maintainers feel free to disagree:
I'm against this. My main concern with this (and other related measures that we've already taken) is that they offer very, very, very little actual improvements to our security and require significant administrative work.

josephlr, 2023-10-12:
Personally, I don't see a lot of value of having a scorecard action. If there's a particular recommendation that folks want to add to this crate, it's probably better to open a specific issue for that recommendation. This will help keep the discussion focused on a particular issue, and reduce the admin work (as @Freax13 mentioned).

closed 2023-10-13 by josephlr